c0pperdragon/LumaCode #1: Request Commodore 128 version

scorpio-ny, 2023-06-24:
Hi @c0pperdragon,
I would like to request a 128 version of this project. I have a question: with the right cable, would this work with the Retrotink or do I have to get a version of the RGB to HDMI upscaler to make use of the video output from the Lumacode?

Thank you!

c0pperdragon, 2023-06-24:
OK, I can design a board to fit the C128, but I can not directly test it, because I don't habe a C128. Still, I am pretty confident that it should work directly because the only real change is the location of the various pins.  My Component Video mod also did work on the C128 without issues.
So I would go forward if you promise to take the first finished board as a beta test.  I would sell it to you for $35, like the C64 version on my store: https://www.tindie.com/stores/c0pperdragon/  
But I will wait to put it up for general sale to get your feedback on how it actually works.

This device will only work in conjunction with a compatible upscaler. Currently this is only the RGBtoHDMI with either an analog board or the new variant that can decode LumaCode and monochrome video signals directly (which I already have on my store). You also need a RaspberryPi Zero to put together the RGBtoHDMI device.

scorpio-ny, 2023-06-24:
Sure, I would definitely purchase the first beta board to test in the the 128. If can send you the payment now via  PayPal if you like. I will need to buy the upscaler from your TIndie store (not a problem). Please let me know on how you want to handle the logistics in terms of payment and shipment. I believe I have a Raspberry PI Zero on hand too. 

Thank you!

c0pperdragon, 2023-06-27:
I finished the design of the PCB for the C128 version. It was actually harder than expected. 
[image]

PCB manufacturing and delivery will take a few weeks. I will come back to you concerining payment and delivery when I have a working board.

Purplegopher, 2023-07-03:
I too would voice my interest in these and am ok with getting a "first version". I have a C128 (non D) to test. I am ok with manual smd rework (though I am not great at that). I have a "normal" RGBtoHDMI analog board (pcb from the repository) available but would order one of your variants, if that they differ electrically.

ami-commo-fan, 2023-07-06:
I am also interested in 128 board if you make few. I was testing the first 128 adatper which you built and very interested in this solution! Thanks for all your work regarding this project!

c0pperdragon, 2023-07-06:
Short update: The PCBs are already on their way.

scorpio-ny, 2023-07-06:
Great! Thank you for the update!

c0pperdragon, 2023-07-16:
The PCBs arrived and I just built the first prototype. On my test harness, it works nicely, and I very much think it will work in a really C128 as well (which I don't have). 

https://github.com/c0pperdragon/LumaCode/wiki/VICIIdizer128-(for-the-C128)

So everyone willing to take the risk can contact me to purchase such a kit at:  reinhard.grafl (at) aon.at

If you need something additional from my store, you an also directly ask and bypass Tindie on this occasion.
https://www.tindie.com/stores/c0pperdragon/

After this beta test succeeds, I will put up the boards for general sale

scorpio-ny, 2023-07-16:
Great! I just sent the email to request to purchase the kit.
Thank you!

Purplegopher, 2023-07-20:
Received mine and did a quick test. Both C128 mode and C64 worked great, no problems observed so far. Easy and clean installation, certainly another great product. Thank you!!

ami-commo-fan, 2023-07-20:
Mine C128 version works also well, after setting the correct values for RGB2HMI and my monitor. Good work!

c0pperdragon, 2023-07-20:
Thank you for this feedback. So that are 2 successes, with 1 to go. It will take a bit for my kits to reach NY. If this is also OK, I will put up the kit for general sale.

scorpio-ny, 2023-08-03:
I finally received mine. And it works great! I am using it  your RGBtoHDMI Mono & LumaCode with the latest alpha firmware.

[image]

c0pperdragon, 2023-09-07:
My dear beta-testers, I have a request for you:
I lately learned that there are actually games for the C128 out there that do make use of the possibility to temporarily switch the CPU to 2 MHz for faster computation. This is only possible in the duration of the screen border, because the VIC-II can make no memory access in this time. As it turns out, my Component Video mod actually had some trouble with this and I will release a fixed firmware soon.
But I have no means to test this (lacking a C128) on the VICIIdizer128 board myself. So please, if you can get "Elite 128" running on your C128, tell me what is happening there. My guess is that it would behave similar, which means, the top half of the screen would show some random flickering. 
Thank you.

scorpio-ny, 2023-09-07:
Would the be the same for the Super Mario Bros. that also use this technique or Eye of the Beholder too?

c0pperdragon, 2023-09-07:
It depends in what the games are doing exactly. Elite 128 is known to have
problems with the other mod.

scorpio-ny, 2023-09-08:
Hi C0pperdragon,

I had done some testing. There are apparently two versions of Elite 128: a version 1& 2. I tested both. Also, I tried testing the other games that I knew used 2Mhz mode: Super Mario Bros. and Eye of the Beholder. For some reason, I could not get Eye of the Beholder to run on my C128. Mario Bros. Was a good way to test because I can manually set turbo 2MHz mode on and off.  I tested by having the C128 connected via two separate HDMI inputs to the same display. One as connected to the Svideo via a Retrotink and the other via Lumacode with your RGBtoHDMI upscaler.  The videos I creates were too big to post here, so I posted them with captions at my site. Here is the link to view them:
[https://www.solo-traveler.com/Technical/Commodore-64/C128-2MHZ-Tests/n-tvp7Xd/](url)

If you have more questions or need more testing, please let me know!

c0pperdragon, 2023-09-08:
@scorpio-ny 
Thank you very much for the test. It indeed seems to be quite the same problem with the 2 MHz mode as the Component Video mod has. I think I can fix this in the firmware, to be delivered with future VICIIdizer128s, but this would not help anyone who already has one. Flashing the firmware is rather tricky and requires extra hardware.
Because I have no test hardware myself, I would like to just do the firmware change and send you a new VICIIdizer128 for free if your are willing to do a careful re-test with various normal games/demos and of course all games/demos using the 2MHz mode.

Purplegopher, 2023-09-08:
I didn't have time to verify scorpio's findings yet but in regards to a firmware update, I guess it would be a similar procedure like for your component mod? I figured out how to upgrade mine a while ago. What exactly would be the HW (and SW) requirements? Just a jtag interface tool and an upload program?

c0pperdragon, 2023-09-08:
From a hardware point of view the firmware upgrade is a bit more involved, but that is not the true reason why I don't want to offer this option. The VICIIdizer firmware is basically my most valuable assert right now and it was quite an effort to develop. So I don't want to give it away as it could easily leak and anyone could build&sell VICIIdizer boards. The board themselves are not so hard to reverse-engineer, but the firmware is read-protected in the chip.

Purplegopher, 2023-09-08:
Ok, I did not think about that, I totally understand that reason.

scorpio-ny, 2023-09-08:
> Because I have no test hardware myself, I would like to just do the firmware change and send you a new VICIIdizer128 for free if your are willing to do a careful re-test with various normal games/demos and of course all games/demos using the 2MHz mode.

Sure, I would be glad to help out.  Let me know if you still need me to send you my email address.

If the firmware fixes work, maybe we can arrange to send back the old ones for reprogramming?

c0pperdragon, 2023-09-08:
No need to send any more mail. I still have your shipping address. Will probably send off a parcel tomorrow.

TheRetroChannel, 2023-10-02:
I think there is something wrong with my 128dizer. My 128D seems to only show an image maybe 30% of the time, and the rest of the time I just get a black screen. Using the reset button on the 128 does not solve it, I need to keep powering it on and off until I get lucky. And on my 128DCR I always get a black screen. 

I can see the 128 is booting normally using the s-video output while this is happening, and the RGB2HDMI is detecting a sync signal from the 128.

Looking at the LUM signal on the scope shows a similar story. When it's a black screen I only get the sync pulses, and when it's working I can see the expected signal.

[image]
[image]
[image]
[image]

c0pperdragon, 2023-10-02:
@TheRetroChannel)
This very much looks like a marginal timing related to writes to the VIC-II registers. Because of the slightly varying length of the pixel clock pulses and the way the VIC then generates the system clock from it, there are 4 possible phase relations in which the machine can power on (a reset does not change this). I guess in the "unlucky" state, the VICIIdizer just misses some important register writes, like the one to turn on the display.

I have been tuning this specific sample point for some time now, but while I hope I have it correct for the C64, the C128 seems to do things slightly differently. Unluckily I have no C128 for development, so I am quite dependent on user feedback.
Can you tell when when you ordered your VICIIdizer128, or more specificially, when did I ship it? I made the last change to the firmware regarding this timing on August, 7th (firmware version 2.2). 
Also, would you be willing to do some oscilloscope measurements on your machine? I would give you specific instruction on what I need exactly.

TheRetroChannel, 2023-10-02:
@c0pperdragon 
Ordered on 6th of September, shipped on the 7th.

Yeah, I can take some measurements. You can email me at markfromtheretrochannel at gmail dot com

TheRetroChannel, 2023-10-25:
@c0pperdragon The replacement 128 version finally arrived, it took quite a while this time. Good news is it now works reliably with the 128 regardless of the clock phase. The only issue I have found is it doesn't play nice with things that use the 2mhz mode. Games like smb64 have a lot of flicker and wrong sprites/colours. Not sure if anything can be done about it but I'm happy to do further tests if you need anything

TheRetroChannel, 2023-10-26:
@IanSB Is there a difference between the Commodore 64 and Commodore 128 LumaCode profiles? I had a quick look and couldn't spot anything. Would it be better to just have a single Commodore 64/128 profile?

IanSB, 2023-11-01:
@TheRetroChannel 
> Is there a difference between the Commodore 64 and Commodore 128 LumaCode profiles? 

No, they are the same, at least initially.
I included separate ones so that different preferences could be set if required. Also it allows separate calibration settings if that is ever a problem.

IanSB, 2023-11-12:
@TheRetroChannel 
beta 61 now available:
https://github.com/IanSB/RGBtoHDMI/releases

closed 2024-01-10 by c0pperdragon