c0pperdragon/LumaCode #2: Feedback Atari version (GTIAdigitizer)

c0pperdragon, 2023-07-17:
Please let us know here about your experiences with the device. Pictures of installation and visual results are appreciated.

scorpio-ny, 2023-08-03:
Hi,

I want to let you that I have received the GTIA digitizer and it works perfectly so far. I am using with your RGBtoHDMI Mono & LumaCode device with the latest posted Alpha software. 
[image]

c0pperdragon, 2023-08-03:
That looks very good. Just to make sure: Both machines (Atari and C128) are NTSC with 60 Hz?

scorpio-ny, 2023-08-03:
Yes, both are working. The only thing I noticed on both machines is that the aspect ratio looks slightly stretched vertically. Additionally,  it looks like color saturation is slightly off from I am used to.

scorpio-ny, 2023-08-03:
No problem! I will review the documentation. There is an important tidbit that I just gleaned from glancing at the documentation.  If one decides to use a PI Zero, it does not support 4K natively. So if one is using it with a 4K display (as I have) it will either display it in 1080P which will produce a softer image  or a special  mode where it supports the display at 4K vertically only. I think that is what I am seeing with my current set up. Could I connect your hat variant to  it to a different Raspberry Pi (i.e 3 or 4) to test this theory?

c0pperdragon, 2023-08-03:
The hat should work with other Pi variants as well. It is mainly the software that needs to all the necessary adjustments.

scorpio-ny, 2023-08-03:
Ok. I have a Raspberry PI 4 but I need to order a HDMI cable for it to test that out. When I get it, I will report back to see if it makes a difference.

scorpio-ny, 2023-08-03:
Oh one more thing. I have other solutions for connecting to the Atari to a 4K display and I I checked the settings for it. They were set to 960x540. So I went in the menu for the HDMItoRGB menu and set that screen resolution manually. And now it looks exactly how I am used to on my particular display.  I will still do some testing with the Pi 4, but this setup so far works perfectly. Thank you for your efforts!

dantaipan, 2023-08-22:
Mine is en route. Do I use the C64 LumaCode profile or is there an Atari-specific profile?

scorpio-ny, 2023-08-22:
Hi @dantaipan ,

You will need to download the latest version of RGBtoHDMI firmware  that has the Atari specific profiles. I downloaded it from this repo [link](https://github.com/IanSB/RGBtoHDMI/releases) 

When I first launched it with my Atari 800XL, the screen dimensions were distorted because it defaulted to a screen mode that stretched it in only dimension . You will need to go in the RGBtoHDMI menu and set your preferred dimensions. I set mine to 960x540. And my system was NTSC.

Hope this helps!

dantaipan, 2023-08-23:
@scorpio-ny got it thanks. Hopefully it will be here before the weekend but starting to look like next week.

r3mowilliams, 2023-08-25:
I've now tried the GTIAdigitizer in two different 800XLs and overall I am so far impressed with it at this point : )

It is not getting along well with the first XL I installed it in.  This XL has a U1MB installed and is otherwise without issue.

https://youtu.be/_Nl4-rC22iE

The second has a UAV mod and nothing else, it is working well in this unit.

I have noticed an issue where GTIA shading is not being correctly represented.

GTIAdigitizer:

[image]

What it should look like:

[image]

c0pperdragon, 2023-08-26:
You could try to fix the first issue by adjusting the settings in the RGBtoHDMI.  Do this by opening the menu (menu button on the RGBtoHDMI), navigate to Sampling and adjust the "Phase" setting. 

I will look into the second issue with the GTIA colors and see if I can reproduce the issue.

c0pperdragon, 2023-08-26:
I could reproduce the effect with the wrong colors in the Ballblazer intro. It seems it is somehow caused by the scanline effect added by the RGBtoHDMI. If you turn scanlines off, everything looks correct (but without scanlines of course).
Please raise an issue at the RGBtoHDMI project for this, so Ian may have a look at it: https://github.com/hoglet67/RGBtoHDMI/issues

scorpio-ny, 2023-09-06:
I went and just posed the issue with the RGBtoHDMI project so it can be looked into.

TheRetroChannel, 2023-09-28:
Just tested this out with the 130XE. Note I have it hooked up directly to the RGBtoHDMI with LumaCode board as the XE doesn't provide an easy way to reuse the RF socket.

[image]

I'm seeing some odd graphical corruption in certain places. For example the Atari Control Program shows a number of issues (colours missing, random junk where there should be nothing) with the LumaCode output that isn't present with the same machine using s-video

[image]

[image]

The drunk chessboard demo also shows some odd junk on the far right of the screen that isn't normally present

[image]

I've played with all the settings in the RGBtoHDMI sampling menu but nothing has worked to solve this.

TheRetroChannel, 2023-09-28:
Ah yes, it is normally cropped out. I adjusted the horizontal position on svideo and it is still there. So nevermind that one. The ACP program is here: https://forums.atariage.com/applications/core/interface/file/attachment.php?id=527461

c0pperdragon, 2023-09-28:
I actually found both problems. The GITA colors appearing a bit too early on the left side was fixed by adding a delay before GITA mode takes effect (this also fixed the grey stripe on the right side of the colors).

The garbage on the right part was caused by the fact that using the GTIA mode turns off the highres mode for the rest of the scan line. So even after turning the GTIA modes off after the color box, everything is shown in low-res mode. I did not know that, so the GTIAdigitizer happily shows high-res pixel instead.

I very much guess that this quirk of the GTIA caused many coders to tear out their hair in frustration. As far as I can see how this specific program works, the developer had to do do quite substantial workarounds (including using player graphics and whatnot) to account for this.

The fixed firmware (1.1) will be used for all future sales of the GTIAdigizier, but I don't quite know what to do with the units already sold. Shipping stuff around the globe is such an effort.

IanSB, 2023-09-29:
@TheRetroChannel 

You can crop the garbage off the image using the crop border(zoom) option in the preferences menu or adjust the capture size in the geometry menu. Would it be better to have the default profile with more cropping or not?

TheRetroChannel, 2023-09-30:
@c0pperdragon 

Yeah not sure what to do about the units already out there. I've played around with it a little more and haven't spotted any issues with anything else. I guess it depends how many units are affected and whether it causes issues with other programs. If it's just that ACP program then it probably doesn't matter.

@IanSB 

I did find a few programs that seem to use the entire visible horizontal area, they don't place any critical information out there (just the borders extend close to the "screen" edge). I'd say leave it as is and people can choose if they want to crop anything

dantaipan, 2023-09-30:
I had not been comparing with s-video and had to spend a lot of time to find something "off" which I guess is good news. I focused on homebrew as it tends to push the limits. Only things I could find were that the sword(?) in Final Assault is missing the texture and the colour on the Arcadia background seems off. 

[image]
[image]

c0pperdragon, 2023-10-02:
The slightly wrong color is a matter of how the palette is set up by the RGBtoHDMI. The GTIAdigitizer only sends color numbers.

This falsely colored weapon is probably implemented using multiple players/sprites layered on top of each other with some extremely unusal priority usage. The priority flags and resulting color merging effects are such an arcane topic that it makes my head spin every time I try to actually understand it. I am very hesitant to change anything here in fear of breaking something else.

dantaipan, 2023-10-22:
Just want to report that v2 is working without any observed issues on the Atari 5200. This shouldn’t really a surprise as it is essentially a 400/800 machine but I thought maybe others would like to know that it is compatible.

scorpio-ny, 2023-10-26:
Speaking of versions, is there any way one can check which version of the firmware is run? If not, would be possible to include some code to allow the end user identify what they are running.

Also, I was thinking. Wouldn’t it be possible to upgrade from the Atari itself by making extra connections to the CPU lines to run a patch? I run a FPGA recreation of the Pokey sound chip and the creator of that project has made so it can be updated from an Atari executable.

c0pperdragon, 2023-10-26:
Sorry, there is no direct way to know which firmware version is on the FPGA. But since I normally only have to make new firmware versions when something does not work, I could write up how to expose the problems and which firmware version fixes that.

About updates: This would only be possible with a dedicated programmer hardware (which indeed could somehow be integrated into the Atari). Also it would mean to give away the firmware binaries, which I don't want to do. The firmware for the various boards took the most effort to make and right now it is nicely protected inside the non-readable FPGA flash. Reverse-engineering the hardware is probably not so difficult and I actually plan to stay in business a bit longer.

Because of this update situation I am willing to give out replacements very cheaply or even for free if the user makes a good case in demonstrating why he absoluteley cannot live with the faulty firmware.   ;-)

scorpio-ny, 2023-10-26:
> About updates: This would only be possible with a dedicated programmer hardware (which indeed could somehow be integrated into the Atari). Also it would mean to give away the firmware binaries, [...]

It is understandable that you want to protect your IP. I was under the impression that you would be able to make the changes with a small patch, not with the entire code base. Thus making the fixes available while still be able to offer some protection to your IP.

> Because of this update situation I am willing to give out replacements very cheaply or even for free if the user makes a good case in demonstrating why he absoluteley cannot live with the faulty firmware. ;-)

So far, I have not seen any show stopping issues that affect the ones I have already acquired and I have been happy with what they can do. It is nice to see that there are options available in case something comes up that severely impacts its usability.

scorpio-ny, 2023-10-29:
> Just tested this out with the 130XE. Note I have it hooked up directly to the RGBtoHDMI with LumaCode board as the XE doesn't provide an easy way to reuse the RF socket.
> 
I was just looking at this with my XE Remake. You may be able to route some cabling outside by going under the ECI connector and use slightly thinner cabling. You will probably need to remove the motherboard to do this.

TheRetroChannel, 2023-10-29:
> I was just looking at this with my XE Remake. You may be able to route some cabling outside by going under the ECI connector and use slightly thinner cabling. You will probably need to remove the motherboard to do this.

I worked out a few options. The cables will just fit past the RF jack if you're going for a solderless solution, alternatively you can disable the composite output and reuse that without needing to solder anything. Otherwise desoldering the centre pin of the RF jack is possible without removing the entire modulator.

dabonetn, 2023-11-04:
I just got my unit, and I've not been able to get a good picture out of it.
I've tried on a atari 130ex with a rambo upgrade, and a stock 800xl, and get the same results on both.
I'm a NTSC user located in the USA.

Here's a short video...
[https://www.youtube.com/shorts/E0mfNoPuuQo](url)

And some stills.

[https://imgur.com/a/64euYRA](url)

c0pperdragon, 2023-11-04:
@dabonet Could you also plug the lumacode signal directly into a composite input on a screen and take pictues/videos of the result. I already have some feedback with simillar symptoms and need to narrow down the possible cause.

dabonetn, 2023-11-04:
https://imgur.com/a/4htvMjI

Looks about the same with the large blocks of garbage and lines, etc..

dabonetn, 2023-11-05:
On just a standard menu screen, the colors (White and Black) are inverted, and when I load something off my sdrive MAX, it really goes nuts.

HDMI out...

https://www.youtube.com/shorts/ElVoqMSyBHg

Monochrome Composite Monitor.

https://www.youtube.com/watch?v=TBi55fdlpkc

Are you testing each unit before shipping?

c0pperdragon, 2023-11-05:
@dabonetn 
I do test all units in my dedicated test rig, but as I recently learned, there are some faults that pass unnoticed. Specifically when there is a bad/missing connection of the GND pin of one of the ICs this may slip through. In a really machine this then causes mayhem.
Could you check with a multimeter if there is continuity between the GND (from the lumacode output header) to each of the 4 logic ICs? (no need to check the central FPGA). The GND pin of these is always located at the bottom right (when looking on the IC so that the numbers can be read).

c0pperdragon, 2023-11-05:
So, there is the problem. This pin needs to be soldered down properly. 
Since I know that my test rig does not show this specific problem, I always do manual test for that.

The question is: Do you have the equippement to repair this yourself? If you want to give this a try and mess up the board, I will send you a new one in any case. But if you succeed this would solve the problem immediately.

dabonetn, 2023-11-05:
Ok, I gave it try and now I have a gnd and a slightly melted socket, and better, but still some random noise that corresponds to audio or maybe bus activity..

https://www.youtube.com/watch?v=LVq9qXIyAEE

So it's still not working 100%.

c0pperdragon, 2023-11-05:
Lumacode on the Atari 8-bit is pushing the limits of what is possible with the transmission circuitry. But you are very close to a stable solution here. 
Try to tweak the following things: When using my cable with the crocodile clips, twist the Lum and GND wires together for best noise immunity and clip he GND wire as closely as possible to the output jack to the ground metal. 
When you use my RGBtoHDMI with the built-in cable, there is nothing to change.
Use the sampling menu of the RGBtoHDMI and try to fine-tune the phase, but also the comparator voltages.

dabonetn, 2023-11-06:
I'm running a new coax cable from the board to the rf jack output on the rf modulator, I've removed the surface mount components from the board and isolated the jack for the output, I've also cleaned the port fully. Using your cable makes no visible difference.

(The coax I used is RG174, and I use it to make cables for the Commodore boards I make and sell, it works very well, but it's not the easiest thing to work with.)

I'll try to find time to try to tweak and fine tune tomorrow. But I'm confused as why you say the interference that's appearing that's synced with the audio is a video issue, There is no video movement on the screen when it's happening.

The C128 version was so easy, this one has been a nightmare. So many hours wasted on defective hardware.

IanSB, 2023-11-06:
@scorpio-ny 
> If one decides to use a PI Zero, it does not support 4K natively. So if one is using it with a 4K display (as I have) it will either display it in 1080P which will produce a softer image or a [...]

Up until now full 4K was not supported on the Pi 4 but I have now added that.
There will be a new beta soon but in the meantime here is an Alpha release with 4K support on the Pi 4:
Alpha61F here:

https://github.com/IanSB/RGBtoHDMI/issues/21#issuecomment-1793938279

scorpio-ny, 2023-11-06:
> @scorpio-ny
> 
> > If one decides to use a PI Zero, it does not support 4K natively. So if one is using it with a 4K display (as I have) it will either display it in 1080P which will produce a softer image or a special mode where it supports the display at 4K vertically only. I think that is what I am seeing with my current set up. Could I connect your hat variant to it to a different Raspberry Pi (i.e 3 or 4) to test this theory?
> 
> Up until now full 4K was not supported on the Pi 4 but I have now added that. There will be a new beta soon but in the meantime here is an Alpha release with 4K support on the Pi 4: Alpha61F here:
> 
> [IanSB/RGBtoHDMI#21 (comment)](https://github.com/IanSB/RGBtoHDMI/issues/21#issuecomment-1793938279)

That is very cool! Are there any performance penalties using 4K output?

c0pperdragon, 2023-11-06:
@dabonetn I am really sorry about this faulty board. When building larger quantities, something always seems to slip through undetected.

It is pretty obvious that there is something that carries over from the audio signal into the lumacode signal to cause interference. I guess the lumacode is already extremely marginal in your case, so the slightest disturbance may just push it over the edge.

According to the datasheet the RG174 as 50 Ohm, which is not ideal for signal as everything is designed around 75 ohm cables. But for short distances this should not be such an issue.

Another thing: Which RGBtoHDMI board are you using? The one from my store with the coax cable permanently attached? Or some other variant. I know there are specific analog  boards that have a slightly worse performance any may cause such an issue. 
@IanSB  knows more about this.

IanSB, 2023-11-06:
@dabonetn

One possible cause of noise like that is the PSU used to power the Pi zero. Some noisy PSUs can interact with the ground paths of the computer in strange ways so try another PSU of a different make if you have one.

Alternatively if your monitor has a USB hub on it, try powering the Pi from the monitor using a USB to micro USB cable although note some hubs on monitors won't power up unless actually connected to a USB host like a PC.

IanSB, 2023-11-06:
@scorpio-ny 
>That is very cool! Are there any performance penalties using 4K output?

The core clock has to run a little faster at 550Mhz rather than 500Mhz so it will run a little hotter than normal but the Pi 4 will run quite a bit hotter than a zero and take a lot more power anyway. The Pi 4 hasn't had that much testing as there was no real reason to use one instead of a zero until now but I'm not aware of any issues at the moment.

The main issue is physically fitting the board onto a Pi 4 which will depend on which type of board you are using. The special lumacode board with captive cable should fit but a standard CPLD board plus analog addon board won't and you will have to use a 40 way stacking extender to clear the connectors on the end of the Pi 4 board.
e.g.
https://www.adafruit.com/product/2223
https://www.adafruit.com/product/1979
https://www.ebay.co.uk/itm/193060049594

dabonetn, 2023-11-07:
> @dabonetn I am really sorry about this faulty board. When building larger quantities, something always seems to slip through undetected.
>
> It is pretty obvious that there is something that carries [...]

The analog boards I've been trying are the one I got from you and one I purchased from retro hack shack a while ago.
I tried 2 other power supplies tonight, one is a motorola charger for my moto phone, another was from a Intel compute stick, both still had the same problems, and actually today it's worse. I really think I did get a lemon unit, and maybe the solder fix you had me try is not the only issue or there is another cold joint somewhere.

Both RGB2HDMI and analog boards give a perfect picture using the C128 board.

c0pperdragon, 2023-11-07:
As you have pretty much tried everything we can think of, I will need to just send you a replacement. Can you tell me your Tindie order number so I get the address from there?

dabonetn, 2023-11-07:
Order Number: 424267

On a side note, I hooked up the TS1000 (zx81) to the analog board last night, and that thing has never looked better!

IanSB, 2023-11-12:
@dantaipan 
@scorpio-ny 
@c0pperdragon 
beta 61 now available:
https://github.com/IanSB/RGBtoHDMI/releases

Imperious685, 2023-11-20:
Just fitted my Rev 2 board to my NTSC Atari 65XE. It has some bad noise issues. I'll have to take it apart again if You want photos of the inside. I previously had a home made S-video mod so the normal video output is disconnected. I am using a rev 3 analog board with my RGB2HDMI and am using the same cable as my vic-20 as It's using the same pins. 5 is lumacode. 3 is Audio. 2

Imperious685, 2023-11-20:
On a separate note in the latest Beta 61 Lumacode 800 doesn't work whereas 5200 does work. Beta 60 Lumacode 800 worked ok.
this was from the first time I tried 61, not after any messing with configs.

c0pperdragon, 2023-11-20:
The Atari 8-bit solution really pushes the limit of what lumacode can do. This noise may well be introduced by your wiring and especially be leaving away the GND return path. 
Try to wire up a dedicated GND from the GTIAdigitizer to your output connector.  Twist this GND wire together with the LUM wire to get best possible noise protection. It may also be a problem when your connector carries composite or chroma so you may have cross-talk. 
To avoid all this, I recommend using a dedicated output RCA jack,. Either a re-purposed RF output or a new one.

IanSB, 2023-11-20:
@Imperious685 
>Just fitted my Rev 2 board to my NTSC Atari 65XE. It has some bad noise issues

That looks like some sort of ground loop or crosstalk issue and I agree with c0pperdragon's suggestion of a direct ground to his board. 
This is almost certainly some sort of hardware issue but if that doesn't help here are a few other things to try:

Go into the sampling menu
Adjust the Y Hi, Y Mid and Y lo DAC settings (only a small amount either way).
Note that final values must be Y Hi > Y Mid > Y Lo

Also try running an auto calibrate sampling phase.

Finally try selecting the RGB version instead of the YUV version of the Atari lumacode profile as these use slightly different settings.
(Revert back to YUV if no difference)

>On a separate note in the latest Beta 61 Lumacode 800 doesn't work whereas 5200 does work

In what way doesn't it work?
I've just tested Beta 61 again with my 800 and it works OK.
Are you using a PAL or NTSC system?
(The Lumacode 800 & Lumacode 5200 profiles are identical)

dabonetn, 2023-11-21:
Replacement unit arrived today, and after a quick install the new unit works perfectly. Thanks!

Imperious685, 2023-11-22:
Thanks for the replies. 1st of all the beta61 config is all good. I overwrote beta60 and that must have messed the 800 Lumacode config. All good with a clean installation.
For the noise issues I have replaced the capacitor on the main board and the ones in the Atari power supply, which cleaned up power supply noise but made no difference to the noise in the video output. 
I tried attaching the ground wire to the Lumacode board which made no difference. I also resoldered the logic ic's on the Lumacode board as there was the odd bit of solder here and there. That made no difference.
There is no crosstalk as there is no signals other than Audio, Lumacode, and Gnd connected to the 5 pin Din.

So I did a bit of poking around. This is a C070025 Rev C NTSC board. When I poke around where U21 isn't and especially when touching W2 which is connected between pins 28 and 16 of the GTIA the noise gets a lot worse. I can't find any problems with the board but will keep looking. this is interesting as according to the GTIA pinout for NTSC models pin 16 is N/C. 
I can't find anywhere what W2 is, perhaps a coil? but that should be L something?

scorpio-ny, 2023-11-22:
> There is no crosstalk as there is no signals other than Audio, Lumacode, and Gnd connected to the 5 pin Din

You may still be picking up cross talk noise from DIN connector. I recommend wiring up from the GTIAdigitizer directly to RGBtoHDMI to see if that makes a difference.

IanSB, 2023-11-22:
@Imperious685 
>Thanks for the replies. 1st of all the beta61 config is all good. I overwrote beta60 and that must have messed the 800 Lumacode config. All good with a clean installation.

Thanks for confirming beta61 is OK.
These updates are always meant to be clean installed on a blank SD card. Sometimes palette and profile names change and if you overwrite you end up with a hybrid install with some files left over from the old version that won't work with the new version.

Looking a bit closer at your noise, it seems to be somewhat similar to dabonetn's noise but more severe as his was only triggered by sound activity.

dabonetn, 2023-11-25:
I spent more time with my defective unit, and I've reflowed every chip and all the legs on the 40 pin socket/pins.
There has been no changes in the symptoms, but it is much much worse on a cold boot vs leaving it on for a long while. 

Cooling off (Freezing) the gtia-digitizer makes the symptoms much worse, but they get less over time, so maybe there were some marginal components in a batch?

Imperious685, 2023-11-25:
I have found the solution to the noise problem. I tried directly connecting the Lumacode output to the RGB2HDMI Analog board. That made no difference, nor did removing W2. As I mentioned before W2 is a Jumper but there is no need for it in NTSC machines as Pin 16 on the GTIA is not connected. 
The Noise did improve as the computer warmed up, but I noticed that if I pressed my finger where L37 and C206 are connected together, or on one side of W2 the noise got a lot worse. Those points are the Oscillator signal to pin 28 of the GTIA. In the Schematics it mentions that L37 can be swapped for a 470 ohm resistor. I desoldered one side of L37 and test soldered a 470 ohm resistor, and that has completely solved the problem. No more noise at all. No need to remove the capacitor either. I just spent half the night playing various games and all good, I can now enjoy the fantastic image quality that the Lumacode board offers. 
    
[image]
[image]
[image]

c0pperdragon, 2023-11-25:
Congratulation that you could solve that!
So the root issue was that the oscillator input to the GTIA was somehow off. Could either have been a non-50% duty cycle or some other fluctuations. It is good to have this now reported here in this issue thread for future reference.

dabonetn, 2023-11-30:
I did this fix on my 130xe with my unit that wasn't working and it fixed it also, so now both units work.

c0pperdragon, 2023-11-30:
As dabonetn also could fix his issue in the same way, this is probably something too look out for on the XE series machines.

dabonetn, 2023-12-21:
FYI, if you want a PAL 800xl in the states with color, this worked for me.

Replacing the Antic/GTIA and crystal while using the GTIAdigitizer gives great video, without needing to add the pal color burst crystal and other components.

CO21698      (PAL XL / XE ANTIC)
CO21698      (PAL GTIA)   
CO16112      (3.546894 MHz XTAL)

Brentarian, 2023-12-21:
Hi,
I cannot get a good picture with the GTIAdigitizer that I received this week.  I have attached captures for you to see.  I tried it in three different Atari 8-bit computers with three different GTIA chips.  The Atari 800 profile was selected and I tried every option I could find.  Any help is appreciated.

The VIC-II-dizer worked perfectly in a C64 on both of the Lumacode RGBTOHDMI units I purchased.

Thanks,
Brent

[image]
[image]
[image]

[43 more replies]