c0pperdragon/LumaCode #3: Feedback VIC-II-dizer (C64 variant)

c0pperdragon, 2023-07-17:
Please let us know here about your experiences with the device. Pictures of installation and visual results are appreciated.

Kayto, 2023-09-02:
I can report success on a C64c shortboard. Looks like there are some subtle differences with the RF modulator and internal layout across similar boards. I needed to cut the wire below the central RF output pin. Might be worth mentioning to try without the wire cut first - given variance noted in the shortboard modulators.
https://twitter.com/AdamT117/status/1697879690738880723?t=YzB2Oth31z6wqpfya8jKCA&s=19

oe7twj, 2023-10-01:
Ich habe Probleme mit meinem VIC-II-dizer in einem C64C.
Der VIC-II ist ein 8565R2.

Das Signal aus dem VIC-II-dizer sieht noch halbwegs ok.
Es ist Monochrome wie beschrieben aber es flackert.
Das Bild ist konstant aber es treten Streifen auf, immer an unterschiedlichen Stellen.

Am RGBtoHDMI Mono & LumaCode ist das Bild ganz ausgefranzt.
Der RasPI ist ein Zero v1.3.
Als Software habe ich die Release Version und die Beta59 probiert.
Bei der Release Version bekommt man gar kein vernünftiges Bild, an der Beta59 siehe Screenshot:

Hat der VIC-II ein Problem?
Was kann ich denn noch probieren?

[image]

[image]

c0pperdragon, 2023-10-01:
Dieses Lumacode-Signal from VICIIdizer schaut ja fürchterlich aus. Kein Wunder, dass das der RGBtoHDMI nicht verarbeiten kann. 
Das kann eigentlich nur durch eine externe Störungsquelle so beschädigt werden. Wie hast du den VICIIdizer mit der Außenwelt verbunden? Über die alte RF-Buchse, wie empfohlen? Hast du vielleicht vergessen, die originale Verbindung vom RF-Modulator zur Buchse zu trennen?  Am besten zeigst du ein paar Fotos, wie die Installation aussieht.

oe7twj, 2023-10-02:
Wie hast du den VICIIdizer mit der Außenwelt verbunden? 
Über die alte RF-Buchse, wie empfohlen? 
Ja genau, aber auch probeweise direkt mit den Krokoklemmen am Stecker zum RGBtoHDMI Mono & LumaCode.
Hab nun eine Krokoklemme entfernt und direkt angelötet (siehe Bild).

Das Bild ist ganz okay über SVideo Kabel an einen HDMI Konverter.

Hast du vielleicht vergessen, die originale Verbindung vom RF-Modulator zur Buchse zu trennen?
Nee, den roten Draht habe ich abgezwickt.

Kann es an der Stromversorgung des VIC liegen?
Oder generell am Netzteil des C64?

[image]

[image]

c0pperdragon, 2023-10-02:
Versuche auch mal, alle verschiedenen Farben auf den Bildschirm zu malen - am besten dicke farbige Balken. Und mache dann ein Foto vom Lumacode-Signal direkt am Fernseher. Vielleicht sieht man ob da spezielle Farben besonders falsch sind.

oe7twj, 2023-10-02:
Vielen Dank für die Hilfe, ich besorge ein Oszilloskop. :-)

oe7twj, 2023-10-02:
So, nun hab ich das mit den Farbbalken mal gemacht.
Links die Farben von 1 bis 16.
Rechts nach Helligkeit sortiert.

Das eine Bild ist vom SVIDEO.
Das zweite vom VIC-II dizer.

[image]

[image]

oe7twj, 2023-10-02:
Update: anderes C64 Netztteil getestet. Das selbe Ergebnis.

Und ich werde versuchen einen anderen C64 als Vergleich zu bekommen.

Die Spannung am C6 ist exakt 3,3V, mit einem Multimeter gemessen.
Die Spannung am

c0pperdragon, 2023-10-02:
Da ist auf jeden Fall mehr falsch als nur eine einfache Störung auf der Signalleitung. Da muss der VICIIdizer einen Fehler haben. Ich werde dir einen Ersatz schicken.
Falls du die Ausrüstung hast und du es versuchen willst, kannst du auch selbst mal die Beinchen der SMD-Bauteile mit dem Lötkolben durchgehen. Vielleicht ist es nur ein schlechter Kontakt irgendwo. Wenns nicht gelingt, bekommst du eben den Ersatz.

oe7twj, 2023-10-03:
> Falls du die Ausrüstung hast und du es versuchen willst ...
Ähm, neeeee! 
Wenn ich da löte ist der sicher vollständig zerstört!
Ich sende ihn zurück, vielleicht kannst du ihn retten ...

>  Ich werde dir einen Ersatz schicken.
Vielen Dank!

Am Oszilloskop sieht man eine glatte Linie bei 3,3V bei DC Kupplung.
Bei AC Kupplung sieht man zackige Kurven.

Keine Ahnung ob 54mV pp viel sind oder wenig?
Die Siebelkos in dem C64 sind original, also sehr sehr alt ...
Er läuft aber absolut zuverlässig! 

[image]

c0pperdragon, 2023-10-03:
Solche kleinen Spannungsfluktuationen sind kein Problem.
Ich kann morgen den Ersatz schicken. Dazu muss ich aber natürlich wissen, welcher meiner Käufer du bist. Am besten schreibe mir eine Nachricht über das Tindie-System.

oe7twj, 2023-10-03:
Super! Ich schreibe gleich. Danke im voraus!

oe7twj, 2023-10-05:
Das Paket ist schon angekommen, das ging aber schnell! 

Der neue VIC-II-dizer funktioniert auf Anhieb wunderbar!

So ein gutes klares Bild hatte ich noch nie bei einem C64!
Es funktioniert perfekt!
Danke, danke, danke!!!

Ich bin begeistert!

c0pperdragon, 2023-10-05:
Super! Das freut mich. 
Wäre interessant, was am alten nicht funktioniert hat. 
Wenn du ihn mir zurückschicken willst, meine Adresse steht ja am Paket. Vielleicht kann ich den ja doch wieder herrichten.

oe7twj, 2023-10-06:
Der defekte VIC-II-dizer ist auf den Weg zu dir. :-)

c0pperdragon, 2023-10-12:
Vielen Dank! Habe das Paket erhalten und den VIC-II-dizer reparieren können. 
Wie sich herausgestellt hat, war bei einem Chip der Ground-Pin nicht verlötet. Seltsamerweise funktioniert das Teil in klaglos meinem Prüfstand. Nur im echten C64 fällt der Fehler auf. Deswegen ist der auch durch meine Kontrolle gerutscht.

Sicherheitshalber habe ich noch meine restlichen Boards durchgemessen, und tatsächlich hatten noch zwei weitere das selbe Problem. War auch schnell repariert, aber ich muss das in Zukunft auch jedesmal testen. Leider fällt mir nichts ein, wie ich meinen Prüfstand verbessern kann, damit das schon dort auffällt.

oe7twj, 2023-10-12:
Oh, ja tut mir Leid, dass ich zu patschert bin, das selbst zu finden und zu korrigieren.
Jetzt musstest du extra einen weiteren zu mir schicken.

Jedenfalls vielen Dank, tolles Projekt, tolles Produkt, ich bin sehr glücklich damit! :)

venice1200, 2023-10-15:
Moin,
ich hab gestern meinen VIC-II Dizer in Betrieb genommen.
Mein C64 hat ein 250469 Board mit einem 8565R2.
Das „eindrücken“ des Zwischensockels ging etwas schwerer vonstatten als bei dem vorher eingebauten Lumafix.
Da ich einen Modulatorersatz eingebaut habe, habe ich mir ein aus einem alten AV Kabel ein passendes Kabel gecrimpt.

[image]

Das Software Setup funktionierte bei mir, mit der aktuellen RGBtoHDMI Version, ohne das ich was einstellen musste.

Mein System braucht an (m)einem Eizo EV2795 (max 2560x1440) oft mehr als 5 Sekunden nach dem einschalten des 64ers bis das Bild stabil angezeigt wird. Und es flackert gerne ein paar mal bis das Bild stabil steht.

Das Bild selbst finde ich genial.

[image]

Danke für diese coole Stück Hardware!

venice1200, 2023-10-15:
Das hier wir mir nach dem einschalten vom RGBtoHDMI angezeigt…

[image]

venice1200, 2023-10-18:
Ist das flackern/blinken beim starten des 64ers normal?

venice1200, 2023-10-18:
Danke für die Rückmeldung.
Das geht auch recht schnell, man sieht halt oft wie das das Bild erscheint, kurz weg geht und erscheint wieder.

Ich schau mal wie ich das Video dazu hier hochgeladen bekomme.

venice1200, 2023-10-18:
https://github.com/c0pperdragon/LumaCode/assets/43762187/709ba0a6-a1f4-49e6-a33f-38bee4a8f56b

venice1200, 2023-10-18:
Eine Verbesserung zwischen Stable und Beta 60 konnte ich nicht feststellen.

c0pperdragon, 2023-10-18:
OK, I see. Its is the same with my RGBtoHDMI an monitor. Maybe Ian can have a look into that. Raise an issue at: https://github.com/hoglet67/RGBtoHDMI/issues

oe7twj, 2023-10-19:
Okay, Super Mario is going on my c128.

Screen is flickering fast .
Sprites seems to be false colors and bad structure?

Here a picture VIC II dizer:

[image]

And one of SVIDEO:

[image]

oe7twj, 2023-10-19:
Sorry false thread

c0pperdragon, 2023-10-19:
ok. so despite my best efforts, the 2MHz hack is just not going to work.
this is a bummer, but I can't help it. thank you for checking

cjmt139, 2023-11-06:
Hi, 

I recently purchased from Tindie/c0pperdragon (sat on the shelf a few weeks until I had time to test).
Setup was a breeze.  Install straightforward.  Case look great.  Could not, really, be much easier.  

**Sadly I am plagued by bad noise to the point that I can't use it.**

Sometimes it starts clean then it deteriorates or starts noisy and stays that way.  Also blacks out quite a bit even with almost static images.  

I have tried it on two two PAL 250425 C64's with 3 different VIC II chips (Pal 6569 R3x2 & R5).  I have tried stable, 59 and 60 releases. Using a Pi Zero 2(w) 2021. I also checked the to C64s clock and calibrated to exact figures. Svideo is rock solid and clean as svideo goes.

Logs and screencaps attached as well as setup pictures below (with r60).

[image]
[image]
[image]
[image]

I have a scope and I am very happy to test or try different settings if you point me in the right direction.  I am guessing setup solid colours and measure the measure levels (ignore back and front porch etc) but I don't really know what the test setup looks like?  Probes on the luma/chroma output obviously but what test signals and how to calc and enter the values in RGBtoHDMI ..??  I could also send scope screencaps too.

Pictures of install:

[image]
[image]
[image]
[image]

A few minor suggestions:

- If you can find one - use a socket with longer legs - my original VIC II socket is a shitty single wipe one - seems a pity to risk changing it so I added a double wipe socket to that socket - otherwise it would not stay in - literally popped out.  I found the machined sockets too risky for trying different VIC II chips - bent some pins while being very careful.
- Print a grommet for the cable entry (I will do when I get time)
- Slope the standoff's slightly that hold the PI with screws - if I tighten the screws then my PI bends upwards and the sockets do not align with the case.

[config.txt](https://github.com/c0pperdragon/LumaCode/files/13261811/config.txt)
[Log.txt](https://github.com/c0pperdragon/LumaCode/files/13261813/Log.txt)

Cheers,
Chris

c0pperdragon, 2023-11-06:
@cjmt139 
When you had your board for a while already, chances are you recieved one of the boards that have a faulty GND connection on one of the ICs. Unluckily, in my test rig, the part will happily run seemingly without issues in this case. Only in a real machine, the problem becomes visible. 
Since I know this I do extra test on all boards, but this was probably too late for your shippment.

Try to confirm this by testing the GND pin of all 4 supporing ICs for continuity (GND is located at the bottom right when looking on the chip so the numbers can be read). If you have the equippement, you could try to fix this issue yourself. Don't fear. If you mess it up, you get a free replacement.

About your suggestions:
The length of the socket legs is indeed a known issue. Unluckily, variants with longer legs are hard to fand and crazily expensive. Also I used a 1.2mm PCB for this specific board revision. With the 1.0mm PCB I use for all other boards and the future VICIIdizer revisions, this issue is slightly improved.
Not sure about fine-adjusting the screw standoffs. In my part, everything sits completely flat. Maybe there was some deviation in your specific print? 
Also the case is a sligthly modified open source model that was intened to fit various RGBtoHDMI variants. So the opening is quite large.

cjmt139, 2023-11-08:
@c0pperdragon
I spent a solid two hours going over each chip (u1 to u4 in detail and looking for anything else that looked wrong), with both an illuminated jewelers loupe (40x?) and some very fine 'needle' test leads on my fluke meter testing continuity.  No legs appear to be disconnected and all show continuity (noting it is very hard to know for sure I got the pad and leg separately).  There was a bit of white 'crud' visible between the legs sort of poking out from under each chip - I cleaned gently with a fine brush and isoprop99% and checked for any visible bridges.  No luck - still noisy as hell.   I have also tried with another Pi Zero W 1.1 - same results.

[image]

c0pperdragon, 2023-11-08:
Oh, there seems to be a miscommunication. I was speaking of the VICIIdizer board that may have missing GND on one of the ICs. Sorry to have you sent off on this tangent. Did not expect the RGBtoHDMI board at all.

cjmt139, 2023-11-09:
Thanks - had a Quick Look this morning before heading to work - there is inneed a disconnected pin - will try to resolder it tonight and advise outcome

cjmt139, 2023-11-10:
Well done - fixed a disconnected pin and found a solder blob on one leg of the main chip (on the VICIIdizer) - it did not look to be bridged.  Has now been stable for 4 hours.

cjmt139, 2023-11-10:
Thanks c0pperdragon,I had success but it was not pretty. The whole chip seems to be sitting up in that corner - actually quite a lot so getting solder to bridge the gap was hard. The chip is off angle and pushed up again its neighbour. tThere were other blobs of solder on or near the main chip. Took me a few goes. I also just touch it and it pops out of the socket (a new one fitted to the existing socket )Also what is the white substance - is it thermal paste or flux?I would appreciate if you would consider sending a replacement (just the Viciidizer) with the thinner board and hence longer legs).I am happy to try to get photos and/or send back if you like - postage from Aus to Europe is bad though.Cheers, Chris

cjmt139, 2023-11-10:
Was trying to PM you but looks like I don't have permission.  While I had success it was not pretty. The whole support chip seems to be sitting up in that corner - actually quite a lot so getting solder to bridge the gap was hard. The chip is off angle and pushed up again its neighbor.  Seems I got unlucky.  There were also few other blobs of solder on or near the main chip which I removed.  Then cleaned flux.  About to give then it worked.

It still pops out of the socket very easily (a new one fitted to the existing socket) - I can't think of a safe way to hold it down.

I would appreciate if you would consider sending a replacement (just the VicIIdizer) with the thinner board and hence longer legs). I am happy to try to get photos to show you (the chip issue).  At these sizes it is so hard to work with - I don't know how you do it.   The HDMI is absolutely amazing on my HP 24 inch 1920x1200.  This is definitely a game changer.

Cheers, Chris

c0pperdragon, 2023-11-10:
Good to hear that you actually managed to repair the part at least from the electrical standpoint. If there is really no way that you can also get a hold on the socket issue, I will send you a replacement. 
This very first batch is really not that great. My soldering improved quite a bit over time, so your replacement should not cause so much trouble.

c0pperdragon, 2023-11-10:
Do you still have the Tindie transaction number, so I can get your address?

c0pperdragon, 2023-11-10:
Before I send stuff around, maybe you could try to use a precision socket instead of the simple socket as is seen in the picture to go between the VICIIdizer and the original socket. I very much think that the VICIIdizer would hold better to this.

cjmt139, 2023-11-11:
Tindie order was #420411

You could be right re the socket - I don't have any as I avoid them like the plague - too much damage getting things in and out and look is non-retro. I go double swipe.

Was as much the soldering as the socket (it is sitting quite high, off angle and pushing against the other chip and was hard to bridge) - surprised if it will stand the test of time.  

Happy to cover postage, pay half or similar - whatever you think is fair - and of course I will report on if thinner board works natively or not.  Otherwise it might be good to ship a machined precision socket with VICIIdizer for others that follow  so it is complete with nothing to buy (or re-price with the longer leg sockets - expensive and hard to find).

Cheers, Chris

c0pperdragon, 2023-11-11:
This first batch of VICIIdizers was really cursed from the beginning. To save a few cents, I used some logic parts from Nexperia (the U2 chip) instead of from Texas Instruments. These proved to be very susceptible to random damage (probably from ESD). So I swapped them out in my whole stock. Unluckily this lead to many replacements to not properly fit any longer and then sit at an angle. Even worse, in my test rig everything worked fine even without proper connected GND pin, so this went undetected. Lessions learned: Only Texas Instruments from now on and individually test all GND pins of all ICs.
Also, I used 1.2mm PCBs because I was concerned about the mechanical stability of a 1.0mm. This then caused the leads to be just a bit too short for comfortable soldering and also as you experienced, the socket can not hold to the board good enough.

Anyway, I really hope this was the last of this particular problems. I will send replacement.

cjmt139, 2023-11-12:
Thanks heaps - great stuff.  This is the most amazing step change for retro c64 - pixel perfect HDMI for Commodore 64.  I suspect the rush you will see will overwhelm your production. Pretty much a must-have purchase for every c64 owner in existence.

BTW I had P?bw?y supply boards, parts and assemble a small run for me (of an RF mod replacement) - it was low cost but boy a PIA.  I wonder what JLP?B assembly costs are like.

c0pperdragon, 2023-11-12:
The rush already has totally overwhelmed my capacity. So users will need to have patiences until I can produce enough of the parts for everyone.

ikrananka, 2023-11-13:
Will this work with a VIC-II Kawari Mini?

c0pperdragon, 2023-11-13:
As far as I know, this does not work together. The Kawari seems to have different signal timings than the real VIC-II and the VICIIdizer cannot really make sense of that.
  
  Reinhard
  
  ----- Urspr&uuml;ngliche Nachricht -----

ikrananka, 2023-11-13:
@c0pperdragon Ah - thank you for clarifying that.  I see you're helping Randy regarding his work on trying to update the Kawari to output lumacode directly.  If that can be achieved then that is likely the best outcome for me as I really want to be able to switch between PAL and NTSC modes quickly and easily.  I'll follow that discussion with my fingers crossed.

cjmt139, 2023-12-16:
I am very pleased to report that the newer version 3 thinner board with slightly longer pins grabs well with the factory original single swipe/wipe Vic II sockets.  The workmanship on the board I received is absolutely amazing - chips are perfectly aligned and it worked flawlessly first go.  See above for install pictures.   Thanks @c0pperdragon

wthorch42, 2024-01-29:
<a href="https://www.youtube.com/watch?v=JQ7hEoClJ3A&t=998s">My install video</a>

emanuelelaface, 2024-05-09:
I have a somehow bizzarre issue. After around 4 hours of usage the screen starts to flicker. I know that is something within the Lumacode board because I tried to extract the video from the composite and is stable, also the menu screen in the RGBtoHDMI is stable on the screen, while the commodore screen blinks. I will try to get a video to document this better but I wondered if someone else experienced the same issue.

If I turn off and on the commodore (even for 30 seconds or so) it restarts to be stable, but I haven't tried for how long it stay stable after a reboot.

c0pperdragon, 2024-05-09:
This sounds a bit like a thermal issue. where some signals that are already marginal are pushed over the border. To better decide if the problem is between the C64 and the VICIIdizer or the VICIIdizer and the RGBtoHDMI, try to feed the lumacode signal directly into a composite TV. If this is stable, you may just need to fine-tune the RGBtoHMI sampling (in the "Sampling Menu").

emanuelelaface, 2024-05-09:
Thank you, I will do some test in that direction and report here.

emanuelelaface, 2024-05-11:
Ok @c0pperdragon I did the test you suggested and the issue is in the VICIIdizer itself.
The flickering happens also on the directly composite input. Also it shows a kind of corruption that is not exactly a lack of sync, it seems a digital issue. I attach here three videos: one is the VICIIdizer connected to the RGBtoHDMI, the second is the VICIIdizer connected to the composite input and the third is the C64 composite output directly to composite in of the screen where you can see that is stable (my first thought was an issue in the Commodore, of course).

VICIIdizer + RGBtoHDMI
https://github.com/c0pperdragon/LumaCode/assets/15322317/1f318575-8a7c-4e52-8bd0-7c07be5c94d5

VICIIdizer + Composite
https://github.com/c0pperdragon/LumaCode/assets/15322317/547830d8-a638-440a-8a8f-e24fd9550995

C64 Composite
https://github.com/c0pperdragon/LumaCode/assets/15322317/0ea4b37a-80a8-4ffd-bcd0-33ba7bf0cd02

c0pperdragon, 2024-05-11:
The issue could be similar to what another user experienced. There the clock generator (some replacement for the MOS 8701) was not powerful enough to drive the VIC-II, the VICIIdizer and an external Ultimate II+ cartridge at the same time. The signal was just a bit to low at times. 
What clock generator is in your machine? Do you have an external cartridge attached? Do you have an oscilloscope to check the pixel clock signal?

emanuelelaface, 2024-05-11:
Interesting. The clock generator is the original 8701, but I have the Ultimate II+ connected, yes. At home I have only a cheap oscilloscope, so I ma not sure I am able to see the pixel clock. Is the 7.88 MHz on pin 21 in the VIC?

c0pperdragon, 2024-05-12:
No, on pin 21 is the PAL subcarrier frequency for color generation. The VICIIdizer uses this only at power-up to determine the VIC-II variant (NTSC or PAL). 
On pin 22 is the pixel clock which drives the digital part of the VIC-II and in consequence the whole machine.

emanuelelaface, 2024-05-12:
After this discussion I opened again my C64 and had a closer look at the board. I noticed that the CT1 trimmer was replaced, maybe it is not set to the proper frequency. So I have to find a better oscilloscope and if I understand correctly I should set the capacitor to see 14.31818 MHz on the pin 8 of the 8701 and consequently 0.985206 MHz on the CPU. I will return to this discussion after I will be able to do these measurements.

It is very possible that if the frequency is not accurately set once the board is warms up it can drift even further and this generates the flickering. I will try to debug this further. It is not a big issue but it is interesting for me to understand better my Commodore. I am not an engineer (I am a physicist) so I am a bit slow on those topics. Thank a lot for your help.

mburgst, 2024-05-20:
In the democompo at X'2024 the Demo 13:37 had problems in the flescos part. See the trivia on CSDB:

https://csdb.dk/release/?id=242855&show=trivia#trivia

As it seems there was a problem with a sprite column which was not shown on the bigscreen via the LumaCode->HDMI connection.

I asked Adam to share this also here, maybe he can shed some more light into the problem as far as he could narrow it down.

c0pperdragon, 2024-05-20:
Members of Fairlight already notified me about this issue. It is really a very weird combination of circumstances that causes this missing sprite. It took me two days to track down what is happening and how to fix it. From now on all VICIIdizers will ship with a firmware version that is compatible with this effect.
Thank you also for taking the time to notify me.

venice1200, 2024-05-21:
Many @c0pperdragon for figuring out this weird issue.

How could you make the FW available to the VIC-II Users?

Is this possible?

c0pperdragon, 2024-05-21:
Updates are only possible by sending me the part for a reflashing. I will do this for the shipping costs alone.

venice1200, 2024-05-21:
Shipping costs to and from you, correct?

All HW Versions can use this update?

[15 more replies]

closed 2024-06-23 by c0pperdragon