c0pperdragon/LumaCode #5: VICIIdizer128 problem

oe7twj, 2023-10-06:
After VIC II dizer work perfectly now in my C64 (many thanks for your support!) I try to get it work also on my C128.

I'm sorry to say, I have problems again and need help.

I did same as for my C64.
- remove VIC-II
- insert  VICIIdizer128
- insert VIC-II
- cut wire from video connector
- connect VICIIdizer128 with video connector

After Power on the LED of the RGBtoHDMI was blinking and after seconds it was always on.
But no picture - totally black.

I tried reconfigure RGBtoHDMI: set it to "Commodore 128 Lumacode" and auto calibrate
But no picture - totally black.

Tried C64 Lumacode again with my C64. 
Bad colors but it shows the start message.
Set RGBtoHDMI to "Commodore 64 Lumacode" and wonderful picture.

====

Now I tried to show the Lumacode as normal Video Signal:

C64: black and white, good screen
C128: black screen

Tried C128 Video from SVideo port: works fine.

====

It seems for me,  VICIIdizer128 always send black screen??

What can I do now?

3,3V is fine on C6.

====

[image]

[image]

oe7twj, 2023-10-06:
Update:

Switching setting "Auto Switch" from "sub profile only" to "off" changes something ...

Now a blinking cursor appears and Power on message in green but black background color.

c0pperdragon, 2023-10-06:
First try to find out which part of the chain is not working. Attach the lumacode signal direct to a compositive video jack of a TV and see what signal comes through (post a picture here).
Try power-cycline the C128 repeatedly and see if the lumacode output correctly comes up some of the time. Other users reported such behaviour which very likely has something to do with how the system clock locks itself to one of the 4 phases of the pixel clock.

oe7twj, 2023-10-06:
Okay, the problem is, maybe I have damaged the VIC-II chip.
There are currently no color, also using svideo, only black and white.

I have removed VICIIdizer128 and inserted VIC-II chip directly as bevore.
But only black and white on screen.
Strange ...

Try to get an new VIC-II chip ...

c0pperdragon, 2023-10-06:
Maybe you have bad contact (or a bent pin) on the chrominance line. When the VIC-IIe is indeed damaged but it only affects the color, it my still be of use to run the system and also make the VIC-II-dizer128 run (once we sorted out the problem why it is not working itself)

oe7twj, 2023-10-06:
Bad contact ... no. I checked this.

Here is a link where is a good tip:
https://www.lemon64.com/forum/viewtopic.php?t=75857

This clock IC 8701 was defective.
I changed this IC and now it is nearly perfectly: 

[image]

It works great, perfect picture quality! :-)

But if I switch off the C128 and switch on again ... black screen.
If I switch to C64 mode, it works fine in blue color.
After this I can go back to C128 mode using reset key, - works great.

But every time I switch off the C128 it starts with black screen again ...

oe7twj, 2023-10-06:
Update:

The real picture is much better than the screenshot above.
This lines in the green are not real, it seems a probmel of my camera.

In real it is perfect!!
Many Thanks again!

Never seen such a good picture from my C128! :-)

c0pperdragon, 2023-10-06:
I have no personal experience with the C128. But if I remember correctly, there is a way to directly powering up the machine in C64 mode. How does it work then?

oe7twj, 2023-10-06:
Oh yes.

You can enter C64 mode by either:
- pressing Commodore key while booting
- enter the command GO64

It works fine in any case.

oe7twj, 2023-10-06:
Update:

Switch off for a short time makes no problem.

I must switch of for a long time to get black screen problem???

Strange

oe7twj, 2023-10-06:
Update2:

Currently it is impossible to produce the problem.
Now it always works fine.
Great.

Maybe cause I have closed the C128 metallic case?

oe7twj, 2023-10-06:
Update 3:

It seems to be a warm/cold problem.
After long time switched off I got again this "black screen".

While RGBtoHDMI has "black screen", also a TV with  compositive video jack see nothing (all black).

After a while the green letters are flickering and a few seconds later the picture is clear.
After RGBtoHDMI works fine, also a TV with compositive video jack see black and white screen, same as with C64.

===

The SVIDEO output always work fine, from the beginning on, independant if C128 is cold or warm.

c0pperdragon, 2023-10-07:
It is very likely that your problem is identical to what a user from Australia experiences. Without a C128 of my own it is pretty difficult to get to the root cause of if, but I guess the oscilloscope readings from this Australian person were conculsive enough to try a solution. I already sent a part with updated firmware but it will probably take some time until I get feedback. 
I propose to wait for this and if it indeed fixes the problem for him, I would send you a replacement too.

I am sorry about all this hassle, but the whole Lumacode project is still in its infancy and there are many problems to fix.

oe7twj, 2023-10-07:
>  with updated firmware

Wonderful!

>  I would send you a replacement too

Hmmm, it would be cheaper if I could do firmware upgrade by myself.
What do I need to do a firmware upgrade?

> I am sorry about all this hassle,

No problem!
It is a hobby no productive system.

I'm very happy with your support!

best regards
Tommy

c0pperdragon, 2023-10-07:
Firmware upgrades would require a dedicated programmer, but more importantly it would mean that I have to give out the firmware binaries, which I don't want to do. These are currently my most valuable assets, so I am a bit hawkish there.

c0pperdragon, 2023-10-17:
I really need feedback on the 2.4 firmware, so I just sent you a new VICdizer128 straight away. Please let me know how this works.

oe7twj, 2023-10-17:
yes of course, will send you feedback immediately after VICdizer128 is here.

oe7twj, 2023-10-18:
Feedback:

perfect!

Opened my c128 and changed VIC dizer 128.
Switch on c128 and immediatly a perfect screen on my TV.

Thank you very very much!!
I'm very happy!

Will send back the other VIC dizer 128 at friday.

best regards
Tommy

c0pperdragon, 2023-10-18:
YEESSS!!!
Finally a working solution. 

There would be only one additional thing to check here:
Some programs do make use of the C128's 2MHz mode to speed up computations a bit. This confused my older firmware so some things on the screen did flicker.
The two uses I know of are: Elite 128 and Super Mario in turbo mode.  If you could test these two games, this would also be great. I don't know where to download them, but they should be fairly easy to find.

oe7twj, 2023-10-18:
Ähm, VIC-II disable screen while CPU is in 2 MHz mode.

Have tested, this works fine without problems.
Screen is green while in 2MHz mode (the border color is everywhere).
After return to 1MHz the screen is as bevore.
No flicker no problem, it works as designed.

Will try to get this games.
But I think it is impossible a game uses 2 MHz mode, cause screen will be disabled and so a game makes no sense?

c0pperdragon, 2023-10-19:
These games are actually using the 2MHz mode only during the blanking interval to do some computations faster. The total speedup would be about 15% or so. But when there are write accesses to the VIC-II registers during this interval, this was not handled correctly in earlier firmware versions, but should be now.

oe7twj, 2023-10-19:
I cannot find Super Mario 128.

But Elite 128 seems to run fine, it is playable.

Grafics is fine, but there is a problem with chars.
Maybe a problem of the CHAR ROM at 2MHz? 

[image]

[image]

c0pperdragon, 2023-10-19:
ok. so despite my best efforts, the 2MHz hack is just not going to work.
this is a bummer, but I can't help it. thank you for checking

c0pperdragon, 2023-10-28:
Thank you, I received your returned VICIIdizer128. I can re-flash it now for sale.

oe7twj, 2023-10-29:
VICIIdizer 128 works perfectly now since your last update.
Now it starts immediatly as VICIIdizer 64 do.

It would be nice to have a 80 char solution also.
Yes I know, there is already a RGBtoHDMI solution for EGA working with c128.
But your solution ist so simple to install and perfectly to use.

c0pperdragon, 2023-10-29:
The VICIIdizer is so easy to install when you re-purpose the RF output jack. If you want to have this a second time, you would need to install an extra jack. Also, the 80 columns mode would need twice the data rate which is actually a bit too much for lumacode. So I am afraid Lumacode is not really a good option for the second video port.

paulhocker, 2023-11-04:
@c0pperdragon i assume this new fix will be in the latest batch of c128dizer when you get a new batch. I too am experiencing the same thing as @oe7twj -- it is only slightly annoying ;) == any word on when the next batch will be available -- I have a c128D that needs some pixel perfect love!

c0pperdragon, 2023-11-04:
Yes, of course I will use the latest firmware for all new builds.

closed 2024-01-10 by c0pperdragon