c0pperdragon/LumaCode #8: Request for TED version

NightriderC64, 2023-11-09:
Hi @c0pperdragon,
I would like to request a TED (C plus 4/C16) version of this project - "TEDdigitizer"? :) It would be great to have it for third I think in terms of popularity Commodore's 8-bit platform and great to see its pixel perfect HDMI 121 colour picture. 
I can be a tester or try to support this fork in other ways.

ps. Great KUDOS for your work!

c0pperdragon, 2023-11-10:
Sorry, but the bit rate needed for this number of colors and sreen resolution is beyond the capabilities of the lumacode system.
The Atari 8-bit computers that need even less are already on the limit and causing issues on some machines.

NightriderC64, 2023-11-14:
This is fully understood but pity anyway...

closed 2023-11-14 by NightriderC64