c0pperdragon/LumaCode #9: Request for Vic-20 NTSC

jc13git, 2023-11-13:
Hello @c0pperdragon,

I saw on your Tindie store that the Vic-20-Dizer is only PAL for the moment and wanted to ask if an NTSC version was in the works.

Thanks for your hard work on this project!

c0pperdragon, 2023-11-13:
Yes. Luckily one user is so kind to borrow me an NTSC VIC. Once I have this I will try to make my board work with this also.

jc13git, 2023-11-13:
That's great news - one more question if you don't mind.  I just recently ordered a VicII-dizer and a Vic20-dizer (PAL) from your Tindie site.  I would like to put the NTSC code on it when you have it working if that's possible -  I'd be willing to help with testing if you need it.  Thanks again.

c0pperdragon, 2023-11-13:
No, sorry. 
The firmware that needs to be flashed onto the part is my most valuable asset and strictly confidential. No way to protect my IP in any other way.

jc13git, 2023-11-13:
Understood - thanks!

closed 2024-01-10 by c0pperdragon