c0pperdragon/LumaCode #10: Using Lumacode with Vic2kawari

paulhocker, 2023-12-16:
I am interested in getting a Kawari so that I can have both NTSC and PAL. Unfortunately the boards do not fit in well with my c64c short board. Do you think it is possible that your vic2dizer would work with the kawari?

http://accentual.com/vicii-kawari/

Thanks again for great product!

c0pperdragon, 2023-12-16:
Unluckily the Kawari is not compatible with the VICIIdizer. It uses different access timings than the origianl VIC-II and my board cannot make proper sense of what it sees on the signal lines in this case.
BUT: the maker of the Kawari is actually working on a way to directly generate lumacode from the Kawari itself. There is a discussion thread on this topic: 
https://github.com/randyrossi/vicii-kawari/issues/1
If he succeeds, this would also serve your purpose.

paulhocker, 2023-12-16:
Oh wow! That is actually so exciting to hear. It is so wonderful to have such smart people doing these things today!!

closed 2024-02-29 by c0pperdragon