c0pperdragon/LumaCode #11: Feedback ZX Spectrum version (ULAdigitizer)

c0pperdragon, 2024-01-10:
This thread is intended for feedback on using my ULAdigitizer.

wthorch42, 2024-01-29:
I successfully installed and tested the ULAdigitizer on a Sinclair ZX Spectrum 48K rubber-key with an Issue 2 board.  Installation was a little fiddly, having to wire between pads on the ULAdigitizer and the ULA pins, but went smoothly.  I spoke with c0pperdragon about that and, in conjunction with some other feedback he'd received, he's considering changing the board design to "castellated", where the board would fit over the pins of the ULA and then you'd just flow solder into the appropriate vias.

I had to tweak the clock trimmer pots on my spectrum to get the clock frequency dialed in correctly to match what's expected by the digitizer, but I think that was specific to my Spectrum, and shouldn't be the case for most.  As I understand it, Issue 3 and later Spectrums don't even have the trimmers.

You'll need at least the <a href="">ALPHA 63A</a> firmware release for your RGBtoHDMI to get the correct Spectrum LumaCode profile.

You can see my installation <a href="https://www.youtube.com/watch?v=0MgvN-2ZBc4&t=1411s">here</a>.

spatuluk, 2024-03-31:
I've been trying to get it to work with my issue 2 Speccy, but it's causing it to enter what seems to be a reset loop.  

Sometimes it gets far enough for the Sinclair Research message to be displayed before it resets again.  Without the digitizer, it's fine, and my ZX-VGA-JOY produces a perfect output.  The composite output is awful, so maybe it's related to that, somehow.

c0pperdragon, 2024-03-31:
Could to provide a photo of the install. Maybe I spot something that is not right.

spatuluk, 2024-03-31:
sure..
[image]
[image]

https://github.com/c0pperdragon/LumaCode/assets/38654713/3c58996a-9000-4e93-bee8-ceacee28e9b2

[image]

The composite mod is unpopulated, and I'm just using it as a platform for the phono.

c0pperdragon, 2024-03-31:
Why did you connect everything with these long wires? I would expect this to wrack havoc on the signal integrity. Using just the shortest connections possible is clearly the first thing to try. You see on my installation pictures how I intended the thing to look.
https://github.com/c0pperdragon/LumaCode/wiki/ULAdigitizer-(for-ZX-Spectrum)#installation

spatuluk, 2024-03-31:
I did that initially, but the result was even worse. it still entered a boot loop, but never displayed a white screen. my soldering was not great, though.

I'll give it another try in the correct position.

c0pperdragon, 2024-04-01:
One thing to check is, if the problem is with the ULAdigitizer or with the RGBtoHDMI and its configuration. Try to plug the lumacode signal directly into a TV with composite video input. You should see a stable black&white picture (with vertical stripes encoding the colors).

c0pperdragon, 2024-04-01:
When the board renders the rest of the system instable, there may be something wrong with the board itself, as it is actually designed to only passively listen to the signals with high-impedance inputs to not disturb the host machine.
Maybe there is something wrong with the voltage regulator or the grounding of the level shifters. Use a multimeter to check the continuity from the ground pin of the ULA (pin 40) to each of the GND pins of the level shifters (pin 10 of both the TSSOP-20 ICs). Also check if these ICs get proper 3.3V on their power pin (pin 20).

spatuluk, 2024-04-01:
Ground continuity seems fine, and I'm getting 3.25v on the power pins. I've re-soldered the digitiser under the ULA, but output is exactly the same. Output to a composite monitor was also the same, with the constant resets meaning it never gets far enough to produce a stable picture. 

[image]

c0pperdragon, 2024-04-01:
Looking at the picture, my soldering of U3 indeed looks a tiny bit suspicious. Maybe there is some  bridgeing that conducts just enough to mess with your system, but  not enough to fail in my test harness.

spatuluk, 2024-04-01:
on U2, pins 4, 5, and 6 are bridged, and 9 and 10.
on U3, pins 12, 13, and 14 are bridged, and 9 and 10.

I'm not sure what to look for with the ULA, but I already checked the CLK, which is 3.5 something-hertz.

c0pperdragon, 2024-04-01:
I am not sure about your measurement procedure, but this amount of bridgeing is basically impossible given that the picture looks not so bad. Maybe it is just the best to give all pins of U2 and U3 a short reflow with your soldering iron. Add no extra solder but use some flux if available.

spatuluk, 2024-04-02:
i just went from pin to pin with the multimeter set to continuity. however, it only gave short beeps for the pins, rather than the constant beep of holding the probes together. it's not documented, so i have no idea what that's about.

anyway, i now have an issue 3 motherboard, with a simple AV mod, and i transferred the board to that. it worked fine first try!  :)

I guess something on the issue 2 mobo is barely working, and the digitiser pushed it over the edge, somehow.  

Thanks for all the help, and sorry for wasting your time with my dodgy speccy. If it's any consolation, the output of the ULAdigitizer is amazing!

[image]

c0pperdragon, 2024-04-02:
Maybe it is just the additional power draw of the ULAdigitizer that rendered your issue 2 board unstable.
Anyway, I am glad that it works now for you at least in the other machine. Congratulations.

L1GHTF0RC3, 2024-08-03:
Hi there,
I am having problems getting this to work with an OSSC.
Tried installing on 2 issue 2 boards and now its installed in an issue 3b, but get the same result.Whatever settings i use on the OSSC i cant get the picture correct.
[image]
ULAdigitizer connected to composite input on tv.
[image]
Connected to OSSC no lumacode.
[image]
lumacode on.
[image]
install.

Hopefully im just missing something here.I cant find anyone else that is using ULAdigitizer and OSSC so i cant copy their settings.
Any help would be appreciated,thanks.

c0pperdragon, 2024-08-03:
It looks very much like the OSSC is not sampling the correct total number of pixels per line. I don't know exactly where this setting is found, but it should be set to 448.

This video shows how to change all relevant settings. It is for the C64, but für the Specturm it should be very similar.
https://www.youtube.com/watch?v=6ZZGEm12jUI&t=180s

L1GHTF0RC3, 2024-08-06:
Thanks for the link, i had already seen that video, thats what made me buy the board.
The thing that wasted time for me was the fact that my TV will not seem to accept 384x240 in line5x mode, i had to change  OSSC to 320x240 output.
Anyway after a lot of playing around with various settings i now have a great picture.
Thanks for your efforts in making this board, it produces a bright clean picture.

rjpontefract, 2025-01-26:
I installed a ULA digitizer in an issue 3 48K spectrum and I believe I have a temperature related problem. The installation works OK when first turned on,  both the direct to composite output and the RGBtoHDMI output look fine.  

However, as things warm up, the RGBtoHDMI loses sync and the direct to composite also has sync issues.  Using the UHF modulator with a composite mod still looks OK.

I used freezer spray on the ULA and IC14 and this seems to resolve things for a short while, although I need to get a more accurate freezer spray to be more specific.

Is there anything known that drifts with temperature that could cause this and could you offer any help to narrow it down to a specific component?  I have an oscilloscope available to check waveforms and frequencies, but I don't have a working example for comparison.

Here's a photo of the installation in case I did anything wrong.

[image]

And here's a snapshot of the direct to composite output when the problem occurs:

[image]

Any help would be appreciated.

c0pperdragon, 2025-01-26:
When the original signal is still working and the machine also runs without
issue, there should be nothing wrong with the clock speed and other signals
as such. But maybe the signals are getting just a bit too weak to be
reliably detected the bhe ULAdigitizer.
Please check with the oscilloscope how the relevant signals look. These are
specifically:
ULA pin 1: CAS
ULA pin 2: WR
ULA pin 32: CPU
ULA pin 33: 33
Best to post pictures of all 4 signals.

c0pperdragon, 2025-01-26:
I just rechecked my implementation, and the only relevant signals concerning synchronization are actually CAS (pin 1) and CPU (pin 32). No need to probe the other pins for this purpose.

rjpontefract, 2025-01-26:
Thanks for replying.  I checked out those signals with the scope and these are the results:

Pin 32 looked the same before and after freezer spray:

[image]

Pin 1 before:

[image]

Pin 1 after applying some freezer spray

[image]

Pin 1 longer timebase

[image]

And pin 1 after warming up again

[image]

The CAS signal on pin 1 looks like the cause of the sync issues.  Any ideas what would cause that to happen other than a faulty ULA?

Thanks again for helping out.

c0pperdragon, 2025-01-26:
I am not sure what is going with CAS. It looks like it has some very short
glitches that causes problems for the ULAdigitzer but not the overall work
of the ULA. So it can not really be said that the ULA is faulty, but it
behaves a bit unexpected.
Unfortunately I don't know now how to fix the issue. Maybe using a very
small capacitor (10pF?) to GND could suppress the glitch, but this could
cause all sort of additional troubles. But I don't have much hope here.

rjpontefract, 2025-01-26:
Thanks for your thoughts, it's appreciated.  I think I'll try one of the modern ULA replacements to see if that helps.  Do you know if the digitizer plays nicely with them?  Otherwise I'll try getting hold of another Spectrum and swapping ULAs.

c0pperdragon, 2025-01-26:
I can't say if ULA replacements will work, but there is actually a good chance. The memory access pattern should be quite identical.

c0pperdragon, 2025-08-02:
> Hello,
> I am using an ULA replacement vLA82 (
> https://www.vretrodesign.com/products/vla82-spectrum-48k-ula-replacement
> ) on a brandnew issue 3 board. I am not getting a stable picture. If I swap
> the vLA82 with a legacy ULA, everything works just fine. Is there any
> chance to get the digitizer to work with the vLA82?

Hi!

The ULAdigitizer relies on the signal timings of the original ULA to figure
out what is going on. It seems the vLA82 may have a different access
pattern to fetch data from RAM, so the digitizer does now know how to
interpret this. I am sorry, but I can not support this.

Maybe the maker of the vLA82 would want to provide lumacode directly. This
should not be that difficult as the digital color data is already available
inside the FPGA.

rjpontefract, 2025-09-22:
> I am not sure what is going with CAS. It looks like it has some very short
> glitches that causes problems for the ULAdigitzer but not the overall work
> of the ULA. So it can not really be said that the ULA is faulty, but it
> behaves a bit unexpected.

Hi!  I finally got hold of another working issue 3 Spectrum and tried swapping the ULAs.  Unfortunately, I still had the same problem with the CAS signal causing sync issues.  Thinking it might be something else, I moved the ULAdigitizer to the other Spectrum and had the same issue with either of the ULAs.

Did I just get unlucky that two issue 3 spectrums would have the same problem or is there a possibility that the ULAdigitizer is causing this?

rjpontefract, 2025-10-31:
The replacement arrived, thank you very much.  Both Spectrums still have the same problem with the new board, so I’m on the lookout for another Spectrum with a different issue board to try.  Should I return the board to you as the original one is most likely fine?  It has been soldered into two Spectrums though, so it’s not like new.