c0pperdragon/LumaCode #13: Feedback TMSdigitizer (for ColecoVision, MSX>)

c0pperdragon, 2024-01-10:
This thread is intended for feedback on using the TMSdigitizer.

IanSB, 2024-01-22:
I've been testing a prototype of the TMSdigitiser with a TI99/4a (50Hz version)
I chose to re-use the existing YUV video connector to output the lumacode signal rather than drill a hole in the case for the phono connector but as there aren't any spare pins I re-purposed the +12v output on that connector which normally powers the external modulator so it is sort of re-purposing modulator functionality in a similar way to re-using the modulator phono.

The 60Hz NTSC version does have a spare pin on its video output connector so that could be used on those machines.

This shows the board removed from the case:
[image]
The green and yellow components to the right of the video connector (near the centre of the photo) are an inductor and capacitor associated with feeding +12v to the video connector.

Both of those components should be removed and the LUMA and GND signals should be connected to the holes vacated by the capacitor: (Brown = GND, Red = Luma
[image]
Do not connect anything to the holes vacated by the inductor as that will connect them to +12V!

There is a large heatsink spacer block bolted to the metal shield which is fortunately the exact height of the lumacode board 
[image]

This shows the heatsink block removed:
[image]

This shows the shield re-installed with heatsink compound spread on the top of the TMS video chip coming out of the bolt holes showing good contact with the shield for heatsink purposes.
[image]
You might be able to omit the sheild and use separate stick on heatsinks instead but I decided to try it this way first.

This shows the lumacode output on the left and the YUV output on the right which I used to check the colours etc. 
[image]
The TMS chip's YUV output can work directly with an analog YUV RGBtoHDMI but it is not stable as the thresholds are too close together and require ajustment as the computer warms up, in contrast the lumacode output is stable.

Screencap from lumacode output on a 1080p monitor:
[image]

wthorch42, 2024-01-29:
I tested the TMSdigitizer with a 60Hz NTSC TI-99/4A and also got good results.  You'll need at least <a href="https://github.com/IanSB/RGBtoHDMI/issues/26#issue-2094944560">ALPHA 63A</a> firmware on your RGBtoHDMI.  Geometry seemed off a bit, so I tweaked it in the profile.  You'll need to dial it in to your own preference.  I wired my output to the +12V pin the same way Ian did above.  I did have to desolder and resolder a clock crystal next to the 9918 to make the digitizer board fit into the socket, but, depending on how the crystal is positioned, you may or may not need to do this.  You can see my install <a href="https://www.youtube.com/watch?v=0MgvN-2ZBc4&t=2376s">here</a>.

IanSB, 2024-01-29:
@wthorch42 
>Geometry seemed off a bit, so I tweaked it in the profile.

Can you post your saved 60hz TI99 profile with the geometry changes. This will be in /saved_profiles on the SD card.

wthorch42, 2024-01-29:
@IanSB 

Hi, Ian.  I can certainly do that 
[TMS99XX_LC_60Hz.txt](https://github.com/c0pperdragon/LumaCode/files/14088734/TMS99XX_LC_60Hz.txt)
, but I want to point out that I'm not very good at setting up profiles for the RGBtoHDMI.  I can't guarantee that I didn't screw something else up in the process.  

Here is a screenshot of the composite output through my RetroTink 2x:
[image]

Here's the adjusted LumaCode version:
[image]

And here's a side-by-side comparison:
[image]

IanSB, 2024-01-29:
@wthorch42 
I think your vertical centreing is not adjusted correctly as there is a blank line under (C)1981 TEXAS INSTRUMENTS so programs that make use of that line as well as the top one will not look vertically centred on the screen.
[image]

CrossBow777, 2026-05-08:
> U4 is also on the TMSdigitizer board itself.

Is that covered up by the VDP? I believe the resistors you are talking about are able to be accessed from the output edge side of the digitizer but aren't all of the ICs covered up with the IC in place?