c0pperdragon/LumaCode #14: Feedback VIC 20 variant (VIC20digitizer)

c0pperdragon, 2024-01-10:
This thread is intended for feedback on using the VIC20digitizer.

wthorch42, 2024-01-29:
The NTSC compatible VIC20digitizer now seems to work well.  Tested with unmodified VIC-20, 6560-101 VIC.  Firmware v2.1 works with no cartridge, Omega Race, Mole Attack, Super Smash, Turmoil and the Penultimate+ cartridge.  You'll need at least firmware <a href="https://github.com/IanSB/RGBtoHDMI/releases">BETA 62</a> on your RGBtoHDMI to get the correct profile.

I 3D printed a port cover bracket for my 2-pin (power) VIC-20 and drilled a hole for the RCA jack rather than doing a permanent case mod and that worked well.  Other than that, installation is quick and easy.

IanSB, 2024-01-29:
@wthorch42 
I replaced the second video output on pin 5 of the VIC20's video connector with the lumacode signal. I think this output is normally connected to the external modulator so it effectively repurposed RF connectivity. This required a track cut on my board but some early VIC20 boards may have a component feeding that signal to pin 5 which could be removed instead

The main video output on pin 4 which is on the same pin as the original C64 still works for direct composite connections.

[image]

[image]

wthorch42, 2024-01-29:
> I replaced the second video output on pin 5 of the VIC20's video connector with the lumacode signal. I think this output is normally connected to the external modulator so it effectively repurposed [...]

Nice.  I didn't go that route because I'd have to make a new pigtail.  I hate soldering DIN plugs and RCA jacks.  Had to do it for the TI-99/4A since I copied you and cannibalized the +12V pin and hated every second of it.

wthorch42, 2024-02-05:
Thermal drift issue:

I was playing some games on the VIC-20 the other evening and I started to experience some issues after having played for a while (10 minutes or so).

I believe the issue may stem from the signals drifting due to thermal issues, rather than a problem with the VIC20digitizer.  I was able to recreate the problem a couple of days later with the same symptoms after about 5 minutes.

However, after moving the VIC20 back to my lab to try to recreate the problem and record it to send to c0pperdragon, I've so far been unable to get it to do the same thing.  I'll keep trying.

For those of you who may want to use the VIC20digitizer, I recommend heat sinking the VIC chip.  This may be especially necessary on the older 2-pin power models as they use an older style voltage regulator and bridge rectifier that generate large amounts of heat inside the case.  Newer models may be less susceptible, but I'd still heat sink the VIC on those, too.  You may even want to consider installing a small 5v fan inside the case to help dissipate heat.

If I'm able to recreate the issue, I'll send the video to c0pperdragon.  I don't know that there's anything he can do to address the issue, but maybe he can adjust the tolerances in his firmware to allow for a larger range of signals to be interpreted correctly.

IanSB, 2024-02-05:
@wthorch42 

Try using the "(Unterm)" versions of the lumacode profiles (under the RGB CPLD) as they have greater noise immunity.

wthorch42, 2024-02-08:
I was never able to get the issue to repeat while in the lab where I could capture the video, so I just added a heat sink to the VIC chip and will try again in the arrangement I had it when the issue occurred.  I'll also try the (Unterm) version as Ian suggested.  If I still experience problems, I'll let you know.

kimholviala, 2024-08-19:
Just got VIC20digitizer installed into a VIC and it works wonderfully. I did find a bug, tested with two separate VICs: when I insert  the TFW8B Penultimate Cardridge and turn on the computer screen remains black until I hit Menu or Reset button on the cardridge (a bit of hit and miss, sometimes I don't get screen at all). It's almost as if VIC20digitizer can't synchronize until it sees the default screen colors - Penultimate's menu is has black background and border with some text on it. Without the Penultimate Cartridge VIC20digitizer gives me a stable screen every time.

c0pperdragon, 2024-08-20:
This issue with the Penultimate Cartridge indeed is caused by the inability of your version of the VIC20digitizer to synchronize when the machine only ever uses the screen mode with 16-line high characters. I fixed this back in January when also implementing NTSC compatibility and ship the new firmware ever since. 
If you want to have the firmware updated, I would do this for shipping costs alone.

kimholviala, 2024-08-20:
Getting new firmware sounds good to me - the issue is a bit annoying because I tend to keep Penultimate always plugged in. Since GitHub no longer allows private messages could you send me an email to kim (at) holviala dot com so we can discuss about details?

dabonetn, 2025-01-19:
For my install I cut the board trace going to both sides of pin 1 (+5v) on the bottom of the board, installed a jumper to go around the cut, and now I have luma code on pin 1, and most of the video cables already have a connector on this pin. Then I just used a rca female to female on the end of the cable.

dabonetn, 2025-01-19:
Are there any alternate palettes for the vic 20, compared to my NTSC monitor, the HDMI just looks dull and lifeless.

IanSB, 2025-01-19:
@dabonetn 

The current palette was based on values from an emulator but if you can find a different emulator with a better palette I could use those values instead. You can also try using the brightness, contrast, gamma, saturation and tint settings in the palette menu to vary the palette.

dabonetn, 2025-01-21:
Gtkvice currently has 5 Palette files in the current release.

colodore_vic.vpl
mike-ntsc.vpl
mike-pal.vpl
PALette.vpl
vice.vpl

Any chance of just getting these added?

dabonetn, 2025-01-22:
Much better with the vice palette..
There's a actual red and things don't look as muted.

Original Palette.

[image]

Vice Palette.

[image]