c0pperdragon/LumaCode #15: Possible color issue with C128 (flat) NTSC and VICIIdizer128

mrgw454, 2024-02-02:
Hello there.  Using a VICIIdizer128 with a Raspberry Pi Zero 2 W on a C128 (flat) NTSC model with stock RF modulator.  Installed (extracted) Beta 62 (from https://github.com/IanSB/RGBtoHDMI/releases) to FAT32 formatted SD card.  Left everything alone with the exception of selecting the Commodore 128 Lumacode profile and setting the resolution to 1280x1024 (for my Dell 2007FP monitor).  I did not use clips to connect the luma and ground wires (I soldered them).  

I appear to have a color issue and unable to figure out why.  

I've attached a few pictures for your review (including the Color Test profile).

Any help or insight would be appreciated.  

Thank you!

-Ron

[image]
[image]
[image]
[image]

c0pperdragon, 2024-02-02:
I am not quite sure what these pictures show me. I guess the top one is the output of the VICIIdizer128, and second the output from the A/V port? So both show basically the same blue on grey text? Is the problem the difference in exact color values? So like the blue not looking just right?
Or is there something else going on, because you would expect a green color here?

mrgw454, 2024-02-02:
Hi.  The output is from the VICIIdizer only (picture of monitor directly) and the rest are from a capture device from the same output.  I'm trying to figure out what may be causing the blue color issue (instead of the green I would expect to see).  Output from the 8 pin DIN (not shown in any of these pictures) looks fine.  I did try another Raspberry Pi Zero (v1.1 instead of v2) and it shows the same blue color issue).  I'm guessing it may be something I did wrong, but can't seem to figure out what that may be.  Thank you for your response and help.

-Ron

c0pperdragon, 2024-02-02:
For a simple and easy test, just put all 16 available colors on the screen at the same time to see how they look.
You can draw colored letters in BASIC, using the color switch keys (CTRL-1 to CTRL8 and Commdore-1 to Commdore-8).
Maybe I can see an error pattern then.

c0pperdragon, 2024-02-02:
And also tell me which colors should be which.

mrgw454, 2024-02-02:
Here's a color test.  
This image is from the actual C128 (screen capture):
[image]

This image is a picture of the monitor running the came test:
[image]

This image is running the same color test program in Vice:
[image]

Please let me know if you need anything else and thank you!

-Ron

c0pperdragon, 2024-02-03:
After analyzing this image it is pretty clear that the various voltage levels produced by the VICIIdizer are not quite what the RGBtoHDMI expects. Basically all levels are a bit too low. When I see this correctly on your photo this is due to the resistor R4 being 150 ohm instead of the required 180 ohm.  No idea how this got through my test procedure.
The best way to fix this would be to replace this resistor. 
But the easier  way is just to down-adjust the expected voltage levels via the RGBtoHDMI "Sample Menu". The relevant voltages are Y Hi, Y Mid, Y low. The drawback of this is that you need to do this again when you do a re-install of the software.

mrgw454, 2024-02-03:
I really appreciate your time with this.  What settings/values do you
recommend I use for those 3  items you mentioned?

Having to reset those values at each software update is not a big deal
since it resolves the original issue.

Thank you!

-Ron

c0pperdragon, 2024-02-03:
I can not guess which values are correct. You just need to try this until the colors get correct without flickering pixels.

mrgw454, 2024-02-03:
Unfortunately, as soon as I start to lower the voltages (even just a couple
of values lower) the image starts to show flickering pixels and really
degrades (without any noticeable improvement in color).

The Y low seems to be really sensitive to any change at all.

At this point, perhaps I can work with you to get it exchanged?

Again, I really appreciate the help.

c0pperdragon, 2024-02-03:
When you change the values it is very much expected to get flickering pixels until you get over the critical area and then settle on a new stable setting. 
For a solid start before fine-tuning, you need to set Y Hi to the original  value of Y Mid, then Y Mid to the original value of Y Lo and Y Lo a few numbers below its original.

c0pperdragon, 2024-02-03:
If you have a soldering iron, you could also try to replace this resistor. I could just send you a replacement in a simple envelope.

mrgw454, 2024-02-03:
Depending on the resistor type (smt or through hole) I'm happy to attempt a
replacement.  I do have a soldering iron and perform light duty work.  I
could probably source the resistor locally rather than have you mail one.

mrgw454, 2024-02-03:
I will do as you suggest and let you know how it turns out.  Thank you

mrgw454, 2024-02-03:
Latest attempt:

[image]

c0pperdragon, 2024-02-03:
Now this looks just perfect. 
Anyway if you want, I will send you a replacement resistor  (a 0805 SMD part) if you thinkg you can swap it out. But please confirm before that this is indeed the wrong part (a 150 instead of the required 180).

mrgw454, 2024-02-03:
I'm happy to source the part locally as I'm fortunate enough to live close
to an electronic parts supplier.  Hate to see you mail something so small
to the U.S. if you don't need to.  Where is that resistor located on the
board?

Thank you

-Ron

c0pperdragon, 2024-02-03:
Starting from the GND output connector, R4 is the 4th resistor in the row.

closed 2024-02-29 by c0pperdragon