c0pperdragon/LumaCode #18: Neuer Lumacode Adapter LED D2 Verhalten

venice1200, 2024-03-15:
Hi,
ich hab meinen zweiten Lumacode Adapter (Kabel mit Kabelbinder fixiert) zusammen mit einem VIC-II Dizer bekommen.

Funktionieren tut er auch soweit, aber im Gegensatz zum alten Adapter (Kabel mit Heißkleber fixiert) bleibt die LED D2 nach dem Start nicht an.

[image]

Beim alten sind im Betrieb beide LEDs an, beim neuen ist im Betrieb nur D1 an.

Ist das normal, hat sich also das LED Verhalten geändert?

Was mir auch aufgefallen ist, bei dem neuen Adapter sind die Jumper J5 und J6 auf der Rückseiten komplett offen.

Viele Grüße
Lars

c0pperdragon, 2024-03-15:
Die Jumper haben in der Konfiguration mit fixem Kabel keine Bedeutung. Bei meinen neuesten Boards mit abnehmbaren Kabel entscheiden die, ob auf den beiden inneren Ringen des TRRS-Steckers GND anliegt oder nicht.

Normalerweise leuchtet D2, wenn ein Ausgabemonitor erkannt wurde. Und D1 leuchtet, wenn eine Eingangssignal da ist und die Ausgabe synchron mit der Eingabe läuft. Wenn D1 blinkt, dann versucht der RGBtoHDMI gerade die Synchronisation herzustellen. 
Wenn alles perfekt funktioniert und D2 trotzdem dunkel ist, dann ist vielleicht die LED selbst oder die Steuerlogik dafür kaputt. 
In diesem Fall würde ich das einfach ignorieren.

c0pperdragon, 2024-03-15:
Nachdem die Steuerlogik für D2 nur aus dem Widerstand R5 besteht, ist der Fehler möglicherweise im Ausgangs-pin vom Raspberry Pi. Sie können mal das Board auf den anderen Pi stecken um den Fehler einzugrenzen.

venice1200, 2024-03-15:
Hi, 
D2 blinkt nach dem einschalten ein paarmal kurz und leuchtet danach durchgehend bis D1 anfängt zu blinken und zu leuchten.

Den Raspberry PI hatte ich vorher auch mit dem „alten“ Lumacode Board verwendet wo beide LED an sind wenn das Bild stabil angezeigt wird.

venice1200, 2024-03-16:
Ich mach das hier erst einmal zu, die LED die LED funktioniert ja.

closed 2024-03-16 by venice1200