c0pperdragon/LumaCode #21: Feedback VIC-II-dizer (C64 variant) Part 2

c0pperdragon, 2024-06-23:


emanuelelaface, 2024-06-23:
Sorry, never thought about a replacement, I am very happy of the VIC-II-dizer, mine was more the curiosity to understand what is going on. For sure I will try with some heatsink to see if I can keep it stable, thanks for your time. I may have also another VIC in another C64 that maybe is less temperature dependent. I will see what I can get.

venice1200, 2024-06-23:
> Sorry, never thought about a replacement, I am very happy of the VIC-II-dizer, mine was more the curiosity to understand what is going on. For sure I will try with some heatsink to see if I can keep [...]

Could you add a picture of your VIC Chip here?

emanuelelaface, 2024-06-23:
> Could you add a picture or at least the Rev and Version of your VIC Chip here for reference if nor already done.

Sure, here we go.
[image]

emanuelelaface, 2024-06-23:
I close here the story saying that with a big heat sink the problem never showed in 4 full hours of usage. So probably this VIC has an high temperature and needs a bit more dissipation to stay within the parameters of the dizer. Thanks to @c0pperdragon for the patience, at least if someone else will experience the same issue knows that is just a thermal problem that can be fixed easily.

Marchalis1967, 2025-01-29:
Hi, good day. Been following the thread. My shipment is already on its way (many thanks btw). For my own knowledge, I want to be certain I am connecting the device correctly. So far from the images it seems straightforward, connect the 2 pins from the device one to the can in the RF modulator, the other to the center terminal of the RF RCA connector. I saw where someone had an issue and you asked if he had disconnected the wire that was native to the center terminal of the RF RCA connector. So my question is, do I have to disconnect that connection before clipping the aligator clip to it?

Thanks in advance.

c0pperdragon, 2025-01-29:
Yes, you need to completely isolate the inner connector of the RCA socket from the orginal C64 circuitry. Only then can it be used to carry the lumacode signal (by clipping the aligator clips to it, or soldering wires directly).

Marchalis1967, 2025-01-29:
Ok, that makes complete sense. Originally I thought the clip was piggybacking. But I see now the intention is to replace completely that connection - which makes more sense (forgive my limited knowledge of electronics).

Marchalis1967, 2025-01-29:
With regard to the video upscaler, this is what I have. Would this work?

[image]

Marchalis1967, 2025-01-29:
Can you give me a link then to what else I would need to get this up and running?
Currently all I have is the C64 and a target HDMI monitor.

Marchalis1967, 2025-01-29:
Oh, so a Raspberry Pi is mandatory then, is that it? I have a Pi3. Assuming that would work?

c0pperdragon, 2025-01-29:
As far as I know, the Pi 3 should also work.
@IanSB Is this correct?

Marchalis1967, 2025-01-29:
Would the Pi (SD Card) have to be flashed with any specific software?

c0pperdragon, 2025-01-29:
Please have a look at the quickstart guide here:
https://github.com/c0pperdragon/LumaCode/wiki/RGBtoHDMI-Mono-&-Lumacode

IanSB, 2025-01-29:
It will work with any Pi except the Pi 5 although you will need a header extension on models like the Pi 1,2,3,4 because the USB connectors will block the 3.5mm socket on the mono/lumacode board.

This is the type of 40 way stacking header extension required:
https://shop.pimoroni.com/products/2x20-pin-gpio-header-for-raspberry-pi-2-b-a?variant=1132812269

 Alternatively the video cable could be soldered to the board instead of using the 3.5mm socket.

Marchalis1967, 2025-02-20:
I want to take this time to congratulate you on this product. I had been going out of my mind trying different options (even different TVs!). But this device works magnificently. Had a few head scratching moments initially but then I was able to understand the menu better and use the required profiles. It's been a long time since I had used a 64 _without any lag_ (as happens on the emulators). Now, just to add to the realism and nostalgia a bit further, I will look at configuring for scanlines.
Nicely done indeed!

VelosCohagen, 2025-03-26:
Hallo in die Runde. Ich habe seit einigen Wochen einen VIC2-Dizer, den ich an meinem C64Reloaded Mk2 in Verbindung mit meinem OSSC betreibe. Der Vic2 ist ein 8565R2. Ich bekomme einfach kein störungsfreies Bild auf meinem Fernseher. Ich habe schon alle möglichen Einstellungen am OSSC versucht, das Bild wird nicht besser.

Leider ist es mir nicht möglich Fotos hochzuladen. Irgendwas funktioniert nicht.

Ich habe vom rechten Bildschirmrand augehend schwarze Streifen im Bild, die flackern, und immer wieder an einer neuen Stelle erscheinen. Die Signalleitungen des Dizers sind über eine RCA Buchse (Anschlüsse gelötet) nach außen geführt. Ich habe schon mehrere teils hochwertige Kabel, als auch einen anderen Vic2 (auch 8565R2) versucht. Ich weiß nicht, was ich noch tun kann. Ich hoffe, dass mir hier jemand helfen kann.

c0pperdragon, 2025-03-26:
Zuallererst kannst du mal schauen, wie das Bild aussieht, wenn man das Lumacode-Signal direkt in einen Fernseher über einen Composite-Eingang reinschickt. Das sollte dann ein stabiles Schwarzweißbild mit schmalen vertikalen Streifen sein. Wenn das mal stabil läuft, dann ist zumindest der VICIIdizer selbst OK.
Leider kann ich dann keine Tipos für den OSSC geben, weil ich mich damit nicht auskenne.

Q47, 2025-04-19:
Seems provided round pin legs don't make good enough contact in the original motherboard chip socket. If I press on VicIIdizer stack, I get a picture. Tested this problem with both Scart output and Lumacode output.

c0pperdragon, 2025-04-20:
If the pins do not want to go into the socket, there may have gotten a bit of solder on them in my production. Please check and maybe you can wipe this away with a soldring iron, if you happen to have one. 
Otherwise it may just be a matter of using a bit more force to push the part in all the way.

Ecki63, 2025-05-08:
Hello everyone,
I am testing my new VIC-II-dizer & LumaCode.
I observe some artifacts thich I cannot get rid of.
They are single coloured dots, randomly appear for 1 frame, and the disapper or appear somewhere else.
Like a glitter-storm. Not VERY noticeable, but you can't miss it either.
I have already tested on different monitors, with different HDMI cables, even with different Raspberry Pi Zeros.
I observe, when selecting a command to reboot, then the image freezes a short time before getting black, and in this frozen time the artifacts are also frozen.
So they are real wrongly digitized numbers, exactly the size of a C64 screen-dot, sitting in the memory.

Unfortunately I cannot check the composite signal because I have no old monitor anymore.

[image]

c0pperdragon, 2025-05-08:
If you can not get rid of this noise be tweaking the voltage threasholds in the RGBtoHDMI sample menu, you could have a noisy power supply for either the Raspberry Pi or the C64. Maybe you can try swapping the power supply for the Raspberry for some other, as very common USB chargers may work.

Ecki63, 2025-05-08:
> If you can not get rid of this noise be tweaking the voltage threasholds in the RGBtoHDMI sample menu, you could have a noisy power supply for either the Raspberry Pi or the C64. Maybe you can try swapping the power supply for the Raspberry for some other, as very common USB chargers may work.

Thank you, c0pperdragon, for the quick reply. So it is good to know that this is a real "noise", not a software problem. I will try to shield the parts, will install ferrite cores also on the power lines and check all the sampling possibilities.

IanSB, 2025-05-08:
@Ecki63

Using ferrite cores on the video or power cables can help but changing the PSU for a different make is the better option, however if your monitor has USB ports, try powering the Pi zero from one of those using a USB to micro USB cable.

Marchalis1967, 2025-05-09:
Can you recommend a link for a PSU I can use? I have vertical waves running
up the display. [image: WhatsApp Image 2025-05-08 at 8.29.32 PM.jpeg][image:
WhatsApp Image 2025-05-08 at 8.30.22 PM.jpeg]

Ecki63, 2025-05-09:
I T   W O R K S   ! ! !
No wrong pixels anymore. What a nice stable picture.

Thank you community     <3

    my test setup

[image]

IanSB, 2025-05-09:
@Marchalis1967 

Your images didn't get included with your post on Github. I don't think images will work when replying via email so you will have to log into github to post them directly.

Vertical waves are not generally a power supply problem but might be fixed by an auto calibration (long press of SW3 or use the menu option)

I use the official raspberry Pi power supply with my systems and sometimes use the USB power out on some monitors.

Marchalis1967, 2025-05-09:
> [@Marchalis1967](https://github.com/Marchalis1967)
>
> Your images didn't get included with your post on Github. I don't think images will work when replying via email so you will have to log into [...]

If I may ask, which button is SW3? Is this the Menu button?

IanSB, 2025-05-09:
@Marchalis1967 

The first thing is to try and determine if the problem is on the commodore 64 side or the RGBtoHDMI side.

Try connecting the lumacode signal out of the commodore 64 directly to the composite video input of a monitor. You will get a black and white image with vertical stripes but check to see if you can see similar faint horizontal bars moving on that image.

Marchalis1967, 2025-05-09:
> [@Marchalis1967](https://github.com/Marchalis1967)
>
> The first thing is to try and determine if the problem is on the commodore 64 side or the RGBtoHDMI side.
>
> Try connecting the lumacode signal out [...]

Was advised this, tried it, problem persisted. I got the same result in BW.
At this point, I'd settle for the elimination of this vertical rolling line. The issue with the Vic II Dizer not functioning I can reluctantly live with.

[image]

IanSB, 2025-05-09:
@Marchalis1967 

Do you get the same effect when connecting the C64 standard composite video output as well?

Marchalis1967, 2025-05-10:
No. It only happens when I connect the VIC II-Dizer setup.
The S-Video solution I'm using shows no interference at all. Even the composite connection doesn't show the issue.

Marchalis1967, 2025-05-10:
Ok so the issue with the vertical waves is solved. I plugged the PSU into a different outlet.
Unfortunately I think I killed the VIC-II-Dizer and/or the PiZero as the power LED does not show it turning on.

IanSB, 2025-05-10:
@Marchalis1967 

I think your problem is an issue with one or both of the switched mode power supplies (the replacement C64 supply or the Pi zero supply) leaking too much current from the AC mains supply into the low voltage DC side. The ripple you were seeing can only be coming from the AC supply as it isn't locked to anything on the C64.
Most small switch mode psus have a low value capacitor directly connected between the AC supply and the low voltage DC side which is used to suppress radio frequency interference. This also results in the AC voltage appearing superimposed on the DC supply but it is normally at such low current that it effectively drops to zero under any load, however if the leakage current is too high it can cause problems. I am not too familiar with 120V AC power but I think it uses split phases so some outlets are on one phase and some on the other so maybe by changing to the other socket you changed to the other phase which would have had an effect on the leakage current.