c0pperdragon/LumaCode #23: VIC20digitizer / RGB2HDMI not working correctly

gorathe, 2024-07-27:
I have an NTSC VIC20, unmodified, with the VIC20digitizer installed, connected to an RGB2HDMI purchased at the same time. The HDMI side of things is working perfectly; however, the output from the VIC20 is having issues. The border colour is wrong—green, not cyan, and the blue text is not blue. Instead, each pixel that should be blue is green to the left if it is the left-most pixel and purple to the right.

I've tried both VIC20 profiles (there is an RGB one and a YUV one, both of which give the same behaviour).

I assume that there is a setting somewhere that I have overlooked - where do I start looking, please?

c0pperdragon, 2024-07-27:
It looks like your NTSC profile might be slightly off, so it can not decode the signal properly. First try to do a clean install (wipe the SDcard first) of the latest RGBtoHDMI software. 
If this does not help, use the "Sample Menu" and set the "Pixel H Offset" value to 1. This should fix the color miss-match on the left and right sides of any changing colors. 
Also try to adjust the sampling voltages "Y Lo, Y Mid, Y Hi" to get correct colors without flickering.

gorathe, 2024-08-16:
Unfortunately, this has been closed prematurely - I have yet to have the opportunity to try the last suggestion. I may be able to get to it over the weekend.

gorathe, 2024-08-17:
Unfortunately, I've not been successful yet. I've attached a few random photos of the setup. I tried a new SD card freshly formatted, and that hasn't helped. I've also tried messing with the Pixel H Offset settings, and I've measured the voltages on the wire between the VIC digitizer and the RGB2HDMI board:
12mV, 310mV, 522mV, 724mV, 936mV.

The DAC levels are configured as:
DAC-A: 0.84V, DAC-B: 0.62V, DAC-C: (Disabled), DAC-D: (Disabled), DAC-E: 0.13V, DAC-F: 0.41V

I've tried two different NTSC VIC-20s, 3 different 6560s (and Victor, youtube.com/@Vicenary), and two different HDMI displays, all have had exactly the same behaviour.

Where do I go from here?

[image]
[image]
[image]
[image]
[image]
[image]
[image]

c0pperdragon, 2024-08-18:
Looking at the display of the oscilloscope, I can say that the VIC20digitizer runs perfectly. So the issue is with the RGBtoHDMI or its settings. 
To make things a bit more complicated there are probably two things wrong here: The sampling order of the sub-pixels is cleary inverted. Therefore you see a differently colored pixel whenever the color changes. This makes is specifically hard to diagnose the other problem by which some sampled levels are detected incorrectly. Please fix the first issue by changing the "Pixel H Offset" in the "Sampling Menu" by just one value (from even to odd or the other way round). I guarantee that changing this setting must do at least something. Please send picture after that.

c0pperdragon, 2024-08-18:
The issue with the missaligned subpixels could also be solved by adusting the "Sampling Phase" in the SamplingMenu. Maybe this even gives a more stable picture.

closed 2025-04-25 by c0pperdragon