c0pperdragon/LumaCode #25: RGBtoHDMI Mono & Lumacode board designs

codewiz, 2025-02-03:
More of a request to clarify documentation than a real issue: I searched the official RGBtoHDMI repo as well as IanSB's fork, but I couldn't find schematics and gerber files for the specialized board described here:

  https://github.com/c0pperdragon/LumaCode/wiki/RGBtoHDMI-Mono-&-Lumacode

In case it's not (yet) released, would it be possible to use this design along with an analog frontend?

  https://www.pcbway.com/project/shareproject/RGBtoHDMI_4_0_Latest_Version_f3429ce7.html

c0pperdragon, 2025-02-04:
The RGBtoHDMI Mono & Lumacode design is closed source. You can instead use  the DIY versions of the CPLD board together with an Analog board. Then you need to wire up the cable like so: 
https://github.com/hoglet67/RGBtoHDMI/wiki/Cables#lumacode--apple-ii--zx80--zx81--uk101--nascom--superboard-ii-or-any-other-mono-computer

codewiz, 2025-02-04:
I would purchase an RGBtoHDMI Mono & LumaCode from Tindie, but the shop says "This item does not ship to Italy."

(My apologies for using a GitHub issue improperly, but I don't know other ways to contact you).

c0pperdragon, 2025-02-04:
Yes, I can use Tindie onyl for sales to the US. Please have a look at my
own store: c0pperdragon.com

codewiz, 2025-02-04:
I placed an order on c0pperdragon.com , thank you! 👍

closed 2025-02-04 by codewiz