c0pperdragon/LumaCode #26: VIC-II-DIZER gives a "no sync detected" message

Piddewitt, 2025-02-08:
moin,

I've received your package and installed the dizer and stuff to my c64c short board.
after the installation and power on no c64 screen is displayed but the message:
"no sync detected" - the auto sync doesn't help

what next?

best
hg

[image]
[image]
[image]

c0pperdragon, 2025-02-08:
There are several possibilites:
1. The VICIIdizer is not firm enough in its socket. This can be checked by looking at the original analog signal generated by the VIC-II (if you have a signal from the original A/V port, all contacts are ok). 
2. You have not cut the wire from the RF socket to the main board, so the lumacode signal can not propery use the RF socket.
3. The white crocodile clip shorts to GND
4. The RGBtoHDMI is not working. To check for this, please try to feed the lumacode signal directly to a composite video input of a regular TV (if you still have such a TV available). You should see a black&white picture with vertical stripes. If you see this, the VICIIdizer itself is OK.

Piddewitt, 2025-02-09:
moin,

I now took another motherboard and checked the VIC before I cut the
modulator wire - the chip and the board are
fully working producing the standard start screen - I now cut the wire and
inserted the dizer - it fitted perfectly into
the socket - the wire is cut - there is no short with ground of the white
crocodile - all results to the same "no sync"

when I directly connected the dizer with an antenna cable it produces a
black screen - so it might be possible
that the dicer insn' working?

what next
HG

c0pperdragon, 2025-02-09:
If you have indeed checked the original analog output with the VICIIdizer already installed and it did work while at the same time, the VICIIdizer shows nothing, it may very well be somehow damaged.
Please post your order number and I will send you a replacement.

Piddewitt, 2025-02-09:
hi,

that's very kind :) the order number was *#222*

I tried to be really careful with the test so it could be quite possible
that the dizer isn't working
do you want to have the old dizer and cables back? I'll send it to you then

all the best
HG

c0pperdragon, 2025-02-09:
Do not send anything back yet. Lets first get everything sorted out.

c0pperdragon, 2025-02-11:
I just sent off a replacement VICIIdizer. Let's hope this works now.

Piddewitt, 2025-02-18:
hi,

thank you , I've received your package.
just to be sure: the red wire is ground, the yellow one is signal?

best
hg

c0pperdragon, 2025-02-18:
The colors of the wires have no meaning besides to tell them apart. The important thing is to which output pin (GND or LUM) you plug the wires to. GND then needs to go the outher shield of the RCA connector and LUM to the inner contact.

closed 2025-04-25 by c0pperdragon

Piddewitt, 2025-04-25:
moin,

I still owe you the money for the 2nd dizer. I can probably send it to your
PayPal account.

thank you for your patience
hg

c0pperdragon, 2025-04-25:
If this was just a replacement for a non-working part, this was free of charge.