c0pperdragon/LumaCode #28: Confirming RGBToHDMI compatibility

otacon6530, 2025-02-19:
I purchased a digital RGBToHDMI years ago here (https://retrohackshack.com/2021/07/20/now-selling-the-rgbtohdmi-adaptor/) from an Aaron Newcomb.

When I saw and purchased the PPU digitizer I recognized the board and thought it was the same one that I currently had, since the form factor, label/name  (RGBtoHD - Hoglet & lanB) is the same. However, after purchase I looked back and found that these were purchased from different stores and under 2 different names. I just wanted to verify if these boards are related and if you think the old version is compatible with the PPU digitizer or if I need to purchase your version to get it to work.

Thank you!

c0pperdragon, 2025-02-19:
No, this board you have is the variant of the RGBtoHDMI with digital input
pins. You need one that has either analog RGB inputs, or at least one
analog input. The RGBtoHDMI Mono & Lumacode from my store would work, as
well as the Analog RGBtoHDMI (for which you also need a cable for lumacode,
as it is not included).
You could also upgrade your existing board with the "Analog board" from
retrohackshack.com, but then you again need this extra cable.

otacon6530, 2025-02-19:
Thank you!

Since your RGBtoHDMI Mono & Lumacode and the Analog board are the same price I will just make a separate purchase via your website. Feel free to combine shipping boxes if that works out cheaper for you. I plan on adding this to my custom nes expansion slot accessory, so I won't be using the cord anyway. 

You saved me a few frustrating hours of my time and weeks of waiting for the required part to come in, so you have my gratitude for being so responsive.

------------------------
Some details if you are interested:

My nes sits on top of a thin 3d printed box that has an expansion slot connector for some of my mods. It just stuff that can be done with the expansion slot right now like the Famicom Disk System audio mod. 

There is some room to the sides of the expansion slot, so I intend to put a thin connector there for the two wires needed for this mod and have the RGBtoHDMI Mono & Lumacode live inside the accessory. I would then be able to have a normal size hdmi connector to the outside of the accessory. Potentially I could buy an hdmi embedder and get sound to work all in a very small foot print.

closed 2025-02-19 by otacon6530