c0pperdragon/LumaCode #30: Feedback STICdigitizer

c0pperdragon, 2025-04-25:
Please let us know about your experience with this device.

r3mowilliams, 2025-06-07:
I've installed the STICdigitizer into my Intellivision III and so far so good with one exception.  While using the Intellivision (Lumacode) palette there is a red/brown hue to the entire screen.  Note that this hue is visible as soon as you select it on the RGBtoHDMI even without a console connected to it.

Colors seem correct (other than the background) while using the Intellivision (Lumacode) palette.  Switching to the other palette available (just plain Intellivision) the red/brown hue disappears (again even without a console connected) - but when the console is using that palette the colors onscreen are incorrect.

First pic Intellivision (Lumacode) palette, second just plain Intellivision palette.  Both captures performed on the RGBtoHDMI itself.

This is using a freshly formatted SD card with beta66 files.

[image]

[image]

c0pperdragon, 2025-06-07:
I did not notice on my computer that the black color was in fact this very very dark brown. I have fixed this palette file in the. You can get an updated version from https://github.com/c0pperdragon/LumaCode/wiki/STICdigitizer-(for-Intellivision)#rgbtohdmi-software

r3mowilliams, 2025-06-09:
Thanks, I am now seeing a black background as expected.

CrossBow777, 2025-12-26:
Thanks @c0pperdragon! Shortly after I posted, I did a file comparison on the palette file I'd been using that is the default since about June and the one sent to me. They were...exactly the same. So I know that the wrong file was being sent to me for my testing. After I confronted the person who asked you to make this about it, they then sent over the correct file.

But all of this did have me learn that the RGB2HDMI apparently creates a bunch of palette files when it loads up and if those files aren't present already. I know this, because I deleted all of the files in the palette folder trying to isolate this, only have find them all back again after the next boot up of the RGB2HDMI. It would be nice if I could prevent it from doing this since most folks aren't likely to need to mess with the palettes for most systems. Because they all show up as they do... It requires you to have to keep selecting through them until you find what you want. And again, it would be nice to only see the palettes for the actual profiles you have present. Because I've been removing most of the profiles from the Lumacode setups I've been doing to try and simplify finding the ones a person wants to use.

c0pperdragon, 2025-12-26:
This would be a feature request for the RGBtoHDMI software: https://github.com/hoglet67/RGBtoHDMI/issues
Unforunately development on this has slowed down considerably in the recent months.

CrossBow777, 2026-01-15:
Not sure this is something specific to the STICdigitizer as I'm guessing it is more to do with the RGB2HDMI software. But I have noticed a bug/issue specific to the Intellivision profiles. This is one of the profiles that makes the main menu fonts and characters larger and stretched out a bit.

As a result of this, I've noticed that pretty much anytime I got into the OSD while on the Intellivison profile and then exit back out. There is always some left over garbage characters that were overlapping on the far right hand side that stay resident on the screen and always there when playing games. The only way to remove these left over characters, is to reset the RGB2HDMI converter.

I've not see any other profile/console combos that cause this to happen. Pretty easy to do though. Just go into a few of the menus when you have an Intellivision profile active and eventually some of the characters will remain showing in the far right side of the screen.

Again, I suspect this is more of an RGB2HDMI software thing but thought I'd mention it here anyway, since I've only seen it happen when using the Intellivision profile.

CrossBow777, 2026-01-29:
Here is something else I've only recently found and noticed in regards to the STICdigitizer and Intellivision profiles. By default the RGB2HDMI will set the 16:9 scaling to enhanced and that is fine. But I think most folks would rather use the Full Width option to see the full borders. 
But...
The image output regardless, still looks a bit more stretched than it normally would. And that leads to the purpose of my question...
The main Scaling option in the Main Menu? By default this is set to to Auto which is some Interpolation method. And this will result in a very sharp image output. But again, the image is a bit too stretch in the case of the INtellivision. Now, you can use one of the Interpolated 4:3 options and that actually makes the image output look more correct in regards to the aspect. However, why is there only 'soft' and 'softer' options for this? If it were an issue with actual scaling and the square vs rectangular pixels issue. Then I would understand that. But when you first make this change with the menu screen active, it actually shrinks the image exact as needed behind the menu and still looks sharp and awesome! But then when you exit out, it resets and you end up with a much software image looking more like one of the RGB setups upscaled. Why isn't there a Interpoloated 4:3 option with sharp? It can be done as again, it does this when you first choose the 4:3 Soft option? 

Is there some reason why that doesn't exist as an option in the RGB2HDMI software? Again, only the Intellivision really gives me a reason to ask about this as it is the one console that seems to show the most stretched look vs others through the RGB2HDMI.

c0pperdragon, 2026-01-29:
I must honestly say, I never went that far into the details of what the
RGBtoHDMI is doing with all this scaling options and whatnot.
All this is always done by the RGBtoHDMI guys, mainly Ian. Unfortunately
this projects seems to have stalled for some time now and Ian does not
respond to any issue reports since October.
My own expertise lies in getting the digitizer boards to work and produce
lumacode. I always prefer integer scaling for my own use for everything,
even if it messes up the aspect ratio.

CrossBow777, 2026-09-08:
Not really an issue with the STICdigitizer itself, but just my current findings with using Intellivision Lumacode on the new Retrotink 6x. At first I wasn't even able to enable the Lumacode option when I had the Intellivision plugged in. Finally discovered that it requires having the Wide Tolerance option enabled in the ADC menu. Once you do that, then you can enable the Lumacode option from the Sampling menu with an Intellivision plugged in.

However, not amount of adjustments that I've been able to change will present a correct image output. The closed I could get with a full screen image is the video below. But it still isn't stable and if you adjust the SoG even one value +\- from this setting, the image will get more unstable or loose sync completely. And nothing I could do ever got the colors to be correct either. BTW, I was using the short 3.5mm to RCA breakout cable with this testing to minimize any signal loss and noise.

So I'm not sure if this is helpful C0pperdragon to forward onto Mike Chi or not...but just letting you know my current results. The Intellivision is one of two consoles that I can test with Lumacode that isn't working through the Tink 6x CE or Tink 4k scalers yet.

https://github.com/user-attachments/assets/9a63149b-a570-45ba-b4a8-d352568ec6fc

c0pperdragon, 2026-09-08:
I don't know how the RetroTINK machines really work internally, but did you try to adjust the values for "Decimation Phase" and "Sub Phase"? I looks like the RetroTINK samples the sub-samples for each pixel on the wrong point, so the colors are all wrong.

CrossBow777, 2026-09-08:
It only provides for 4 options as you can see along the top. And none of the 4 options gave anything close to the correct colors. I'm more concerned with the lack of stability with the image itself. Because as I said, any adjustment to the SoG threshold beyond what I have listed there in either direction will just cause the image to fail completely at that point. Similar behavior happens if you adjust the sub-phase as well. The settings shown in the short video is the closest I can get to any sort of stable output to see anything clearly. Even then it still isn't that stable as you can see.

The only way to change those decimation values, is to modify the .lmc file that is part of the palette profile for the console and change the values of the sampling and decimation there. But I've no idea what or how these starting values were even come up with in the first place to know what I should try as alternates.