c0pperdragon/LumaCode #31: Feedback TEDdigitizer (for C16 and Plus/4)

c0pperdragon, 2025-04-25:
Please let us know about your experience with this device.

c0pperdragon, 2025-08-17:
> Hiya, please may I ask if these are available to purchase anywhere? I
> can't seem to find them on your shop, or on Tindie (apologies if they are
> there, and I am being dumb). I am happy to help with any testing etc. if
> that's still useful to you?

Sorry, this is not ready yet for release, because the RGBtoHDMI does not
support the necessary encoding.
Maybe you can raise an issue in the rgbtohdmi issue thread to make Ian
aware if it. (I myself do not want to constantly nag him about it).
Reinhard

mvbhz, 2026-03-05:
It would be nice to have access to both options (if possible). Perhaps option 1 would be acceptable? If not, option 2 could be enabled.

mvbhz, 2026-03-16:
> I was able to squeeze the data transfer just enough to allow for a 4 pixel
> wide horizontal border and this now works nicely with the RGBtoHDMI. While
> this border is really narrow, it may be enough for programs to convey some
> necessary information
>
> The TEDdigitizer is already available at my store: c0pperdragon.com
>
> have much fun

I have another question, is the TED profile already available or do I have
to download it from somewhere?

mvbhz, 2026-11-05:
The image is fantastic. Just compare a standard analog-sourced image with an image from the TEDdigitizer:

[image]
[image]
[image]

Another fantastic piece of news is that you can obtain a larger vertical screen area by using the following settings in your profile (Saved_Profiles\6-12_BIT_YUV_Analog\Commodore\Commodore_16_Lumacode\C16_LC_312.txt):

sampling=7,7,7,7,7,7,7,1,1,0,1,1,0,0,0,2,0,2,1,0,62,47,256,256,8,30,256,256
geometry=0,57,328,208,328,270,1,1,3,2,21281376,1368,6000,312,4,0,0

Unfortunately, I have not been able to increase the horizontal screen area.

c0pperdragon, 2026-11-05:
There is no way to get more horizontal screen area, as the TEDdigitizer does not send any more of the boarder. This is because there is not enough time per scanline to squeeze more data transfer into it, given the constraints of the lumacode transmission speed.

mvbhz, 2026-11-05:
Yes, I know. But it’s good that you can at least get a larger area in the vertical orientation. That’s already a very good thing!