c0pperdragon/LumaCode #38: NES PPU Digitizer not working with RGBtoHDMI Mono & Lumacode with Audio

lupinthird, 2025-08-25:
Hopefully, I didn't buy an incompatible combination of products!

The PPU digitizer doesn't seem to be working.  I have it soldered onto the solder side of the NES mobo with the correct orientation -- the Luma and GND pins are on the side nearest the power PCB.  Since I wanted to make an integrated HDMI solution, I did solder the Luma and GND wires directly to the 2 pads on the RGBtoHDMI hat -- testing out continuity with a meter before soldering them on.

So on my RGBtoHDMI, only 2 pads were soldered to the audio board, the center pad and left pad I believe.  I'm not even trying to get audio working at this point, I'm just trying to get a working video signal from my NTSC NES.

My NES video output still works if I ignore the Luma code/RGBtoHDMI circuit.  Full disclaimer, I have one of Tim Worthington's NESRGB 4.X PCBs on this build as well.

Okay, probably need to get around to discussing the behavior, huh?

Basically, the RGBtoHDMI seems to be working fine, the menu shows up and buttons work as expected, etc.  However, I'm getting a buzz noise coming out of the board (via HDMI).  No video, no intelligible audio, just buzzing audio and black screen with RGBtoHDMI menu.

I suspect that I'm not doing something correctly with the signals or pads, but since it was basically built already, wasn't sure what else I even needed to try.

Any ideas?  Also would like clarification if audio is even supported yet for the PPU Digitizer.  If not, that's okay, I mainly bought the audio version for other setups I may tinker with soon.

Thanks for such a super cool project!
Chris

lupinthird, 2025-08-25:
[image]

[image]

CrossBow777, 2025-08-25:
The problem is you attached the wires to the audio boards input pads. You have the lumacode wire soldered to the audio input and it needs to be soldered to the other connection on the 3.5mm audio input jack on the RGB2HDMI directly.

That is why you have the buzzing and no video. 

And the audio mixer with lumacode only supports MONO audio so that is why there are just the two connections to the audio board that has been soldered to the top.

c0pperdragon, 2025-08-25:
When soldering the wires directly instead of using a plug, you need to solder the lumacode wire to one of the pins of the 3.5mm connector that can just be seen between the legs of the reset button. Both are connected together, so you can solder to any of them. the GND wire ist OK as it is.

lupinthird, 2025-08-25:
All I can say is _awesome._  Everything looks good now, audio is enabled, couldn't be happier.  Cheers, mates!

[image]

closed 2025-08-25 by lupinthird